earlephilhower/arduino-pico #2: Is it support Arduino IDE in macOS?

Dracanrage, 2021-03-18:
I have do the instruments in README on my Mac, but seems the arduino IDE can't identify the develop board.
[image]
[image]

earlephilhower, 2021-03-18:
I'm not a Mac user, but I will try my best.  Since it is not showing the boards at all, the `hardware` directory may be in the wrong spot.

`hardware` should be in the same directory as your sketches.  Under Linux, that's `~/Arduino/`, on Windows it's `\Users\XXX\Documents\Arduino\`.  There's some place similar on the Mac.

Good luck!

closed 2021-03-18 by earlephilhower

Dracanrage, 2021-03-18:
> I'm not a Mac user, but I will try my best. Since it is not showing the boards at all, the `hardware` directory may be in the wrong spot.
>
> `hardware` should be in the same directory as your [...]

Thanks!
I finally now how to deal with it, I have put it in wrong path, now I can find it in IDE.
But when I try to compile it, it shows errors as below:
```
In file included from /Users/mac/Documents/Arduino/pico_main/pico_main.ino:4:
/Users/mac/Documents/Arduino/hardware/pico/rp2040/cores/rp2040/Arduino.h:28:10: fatal error: api/ArduinoAPI.h: No such file or directory
   28 | #include "api/ArduinoAPI.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

Dracanrage, 2021-03-18:
> > I'm not a Mac user, but I will try my best. Since it is not showing the boards at all, the `hardware` directory may be in the wrong spot.
> > `hardware` should be in the same directory as your sketches. Under Linux, that's `~/Arduino/`, on Windows it's `\Users\XXX\Documents\Arduino\`. There's some place similar on the Mac.
> > Good luck!
> 
> Thanks!
> I finally know how to deal with it, I have put it in wrong path, now I can find it in IDE.
> But when I try to compile it, it shows errors as below:
> 
> ```
> In file included from /Users/mac/Documents/Arduino/pico_main/pico_main.ino:4:
> /Users/mac/Documents/Arduino/hardware/pico/rp2040/cores/rp2040/Arduino.h:28:10: fatal error: api/ArduinoAPI.h: No such file or directory
>    28 | #include "api/ArduinoAPI.h"
>       |          ^~~~~~~~~~~~~~~~~~
> compilation terminated.
> ```

Oh, I find the README now, it really works, thanks!