earlephilhower/arduino-pico #4: Still required to set a serial port?

aallan, 2021-03-22:
> There is automated discovery of boards in bootloader mode, so they show up in the IDE, and the upload command works using the Microsoft UF2 tool (included).

I was still required to set a serial port before uploading code to the board? Upload proceeded by copying a UF2s expected after I'd done that (it works!), but having to set a serial port is going to be pretty confusing, and some users may not have any serial port options offered to them so will be unable to proceed?

earlephilhower, 2021-03-22:
If you manually reboot into UF2 mode, it should just work w/o a serial port (the 1200bps twiddle won't be done).  Is that not the case for you?

I think it may be more of a documentation thing, because after that 1st load the serial port will (should) auto-list and it's no different than, say, an AVR or ESP8266 where you set the port once and it just sticks...

earlephilhower, 2021-03-25:
Just updated the README.md with a note about this.  Think we're good here, thx!

closed 2021-03-25 by earlephilhower

aallan, 2021-03-25:
Oh. I've figured it out. It works out of the box for production Picos, but I was using a pre-launch prototypes which has some differences. That explains that then. Apologies!