earlephilhower/arduino-pico #5: Error uploading:

technoblogy, 2021-03-23:
My application compiles successfully, apart from five warnings:

    /Users/david/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.0.1-base-3a57aed/bin/../lib/gcc/arm-none-eabi/
    10.2.0/../../../../arm-none-eabi/bin/ld: warning: changing start of section .bss by 4 bytes

But then the upload fails with:

    Converting to uf2, output size: 516096, start address: 0x2000
    No drive to deploy.
    An error occurred while uploading the sketch

I was able to successfully upload your **Fade** example. Any suggestions?

earlephilhower, 2021-03-23:
That warning is fine, it's just the linker being chatty.  No worries there.

It seems the UF2 drive didn't get mounted by the OS.

Since the reboot-to-UF2 is actually SW controlled (via SW controlled USB), if there's a hard crash in the Pico, it won't work.  In that case, just reconnect the USB cable while holding down BOOTSEL to ensure the ROM itself enters the upload state.

Macs are also a little tricky w/the serial port controls changing between OS releases.  #2 was able to get things running find, but did not report his OS level.

Another possibility is that the Mac isn't mounting the UF2 volume automatically.  To fix that, I'd need a Mac user's help as it's out of my cross-compiling domain.  There may be a setting in the OS about mounting USB sticks automatically, or there may be a CLI command that can cause MacOS to mount things.  Under Windows and Linux it "just works" so there's nothing special there.

technoblogy, 2021-03-23:
OK, I can now get the upload to work if I unplug and then plug back in with the BOOTSEL button held down.

I now don't seem to be able to get any output on the Arduino Serial Monitor. For example, running your example eeprom_write.

I'm on Mac.

earlephilhower, 2021-03-23:
Macs and serial ports always seem to be troublesome. :(  Some of the same issues were seen on the ESP8266 Arduino port https://github.com/esp8266/Arduino

Do you have the proper port selected in the Arduino menus?  (Would also affect the auto-reset...)

[image]

You can use the `screen` command line tool to check comms.  Close the Arduino serial monitor and then from the command line use `screen /dev/ttyXXXX` where the TTYxxxx is the path shown from the menu.

technoblogy, 2021-03-23:
Yes I think so:

[image]

technoblogy, 2021-03-24:
Closing because there is another issue about the Serial output.

closed 2021-03-24 by technoblogy

me912, 2021-11-28:
I found the solution. Firstly, you push bootsel button and go to arduino ide download upload your code to board and it work.

odehjoyce, 2022-05-25:
I have tried all the solutions stated here and it doesn't work. I am using a windows os

odehjoyce, 2022-05-25:
Finally solved it. Thanks

smarttech24, 2023-02-24:
> I have tried all the solutions stated here and it doesn't work. I am using a windows os

how you solve that