earlephilhower/arduino-pico #6: User with space not working

ArielWolle, 2021-03-23:
On windows I was dumb and have a user with a space in the name and now the file is going to the wrong place. Im sure the package is there
[image]

ArielWolle, 2021-03-23:
its supposed to be "C:\Users\Ariel Wolle\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\0.9.0" but its currently "Wolle\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\0.9.0"

ArielWolle, 2021-03-23:
changed account file name and now works it was definitely the space

earlephilhower, 2021-03-24:
@ArielWolle yes, it's related to a space in the path and missing quotation marks around filenames.  

Can you try #7 out and report back?

If you can't use `git`, then just download the file https://raw.githubusercontent.com/earlephilhower/arduino-pico/40e014419586a29c47ab548dc0ad038c82979436/platform.txt , overwrite the `platform.txt` in the rp2040 directory you were looking at, and restart the IDE (restart required or Arduino will not read the new file).

ArielWolle, 2021-03-24:
yup works perfectly thanks for the fix

PR #7 (2021-03-24): Allow for spaces in paths of the platform/core/etc.
Properly quote all instances of file paths and runtime paths in the
platform file to ensure thathaving a username with spaces in it won't
break the compile.

Fixes #6
commit 40e01441 (2021-03-24): Allow for spaces in paths of the platform/core/etc.

Properly quote all instances of file paths and runtime paths in the
platform file to ensure thathaving a username with spaces in it won't
break the compile.

Fixes #6
closed 2021-03-24 by ArielWolle