earlephilhower/arduino-pico #8: Serial print example

ArielWolle, 2021-03-24:
Maybe im just not trying hard enough but Serial.print("A") with Serial.begin(9600) isn't working. Maybe add an example to show users how.

earlephilhower, 2021-03-24:
@ArielWolle great timing!  I was seeing the port unrepeatably disappear from the Serial Monitor (but always working on standard minicom and hyperterm).

In #9 there is a 1-word change in 1 file you can manually apply.  Can you apply that change on your own and verify it now works?

If so, I can generate a 0.9.1 release so everyone will be updated with the fix.

ArielWolle, 2021-03-24:
Im not getting any serial output can you send me a test file for serial output?

earlephilhower, 2021-03-24:
It could be something driver related, or an IDE Serial Monitor issue.  Can you check the device manager and verify that you see the COM port?  Then, can you give `putty` or another serial terminal a run and see if they can communicate with it?

ArielWolle, 2021-03-24:
Yup works on putty and i put in that 1 word change thanks for helping

earlephilhower, 2021-03-24:
It's looking like an unexplained behavior in the Arduino Serial Monitor.  I can occasionally get it to fail to appear under Win (Linux seems solid) , but when the Win Arduino Serial Monitor doesn't show anything, just running `putty` gives proper output w/no reboot/replug/etc.  Very odd.

I'll do an 0.9.1 release anyway, and see if I can dig some more, or if someone pops in and has some ideas.

technoblogy, 2021-03-24:
I also can't get any output on the Arduino Serial Monitor. For example, running the example **eeprom_write** (I'm on Mac).

I tried the fix **Use CDC class for the USB serial port** but that didn't help.

earlephilhower, 2021-03-24:
@technoblogy @ArielWolle check the 0.9.4 release.  @me-no-dev provided a fix to the config file allowing the Arduino Serial Manager to work properly.  Can you please try and report back?

ArielWolle, 2021-03-24:
Yup works great :)

closed 2021-03-24 by earlephilhower

technoblogy, 2021-03-24:
Yes, the Serial Monitor works for me now too. Thanks!

earlephilhower, 2021-03-24:
All thanks to @me-no-dev!  He manages (among other things) the ESP32 Arduino repo https://github.com/espressif/arduino-esp32/ and was kind enough to point out the issue.

PR #9 (2021-03-24): Use CDC class for the USB serial port
Fixes #8 and some other weirdness seen when going between computers.
Use the CDC and not MISC class to identify the Pico
commit 058e9e24 (2021-03-24): Use CDC class for the USB serial port

Fixes #8 and some other weirdness seen when going between computers.
Use the CDC and not MISC class to identify the Pico