earlephilhower/arduino-pico #17: Nop a issue - Display resources, Ram and Flash in complier

vslinuxdotnet, 2021-03-25:
Hello,
In some future day is possible to add the free/occupied resources at the end of the compiler console ?
Thank you

earlephilhower, 2021-03-25:
It's relatively straightforward and a good project for someone looking to jump in and help.

See how it's done in https://github.com/esp8266/Arduino/blob/master/tools/sizes.py .

You'd also need to find total flash size from the UF2 file using some Python like:
````
with open(uf2file, "rb") as f:
    blk = f.read(512)
    s = 0
    while blk:
        ret = struct.unpack("8i476xi", blk)
        if (ret[2] & 1) == 0:
            s = s + ret[4]
        blk = f.read(512)
    print(s)
````

vslinuxdotnet, 2021-03-26:
OK I make a change in platform.txt in line: recipe.objcopy.uf2.pattern=
I replace the cmd "elf2uf2" to the build.py to process some python code in the middle.
This is a test with a blink code:

[image]

Now lets do some code to get it for memory.

earlephilhower, 2021-03-26:
Cool!  Looking forward to a PR sometime.

I would not print ELF or UF2 file sizes, they're unimportant and not really related to what's in the flash (i.e. ELF has debugging information and UF2 has large overhead for each sector of flash)

vslinuxdotnet, 2021-03-26:
Seams that is not need to invented the Wheel, the Arduino have all things need for that change:
the `compiler.size.cmd=arm-none-eabi-size` exists so....
in file boards.txt add/change this lines for flash and ram:
`rpipico.upload.maximum_size=2097152`
`rpipico.upload.maximum_data_size=270336`

And in file platform.txt add this line in: ## Compute size:
`recipe.size.regex.data=^(?:\.data|\.bss)\s+([0-9]+).*`

Works:

[image]

That's all for today :)

earlephilhower, 2021-03-26:
Great, thanks!!!

PR #31 (2021-03-26): Add project size to compile output
Thanks to @vslinuxdotnet for digging into this!  Adds the compiled
sketch RAM and flash usage to the output.

Fixes #17
commit 6062ba90 (2021-03-26): Add project size to compile output

Thanks to @vslinuxdotnet for digging into this!  Adds the compiled
sketch RAM and flash usage to the output.

Fixes #17
closed 2021-03-26 by earlephilhower