earlephilhower/arduino-pico #20: Board not being detected by the IDE

brandonplank, 2021-03-25:
Hey, I've been pressing down the BOOTSEL button on boot and its still not recognizing my device

I've tried with multiple picos.

Yes, i have installed the module. I am on windows if that helps.

brandonplank, 2021-03-25:
[image]

earlephilhower, 2021-03-25:
Looks like you're running the beta 2.0 version.  I've been unable due to some library issues on my machine, but I know others are using it under Windows fine ( https://www.tomshardware.com/how-to/program-raspberry-pi-pico-with-arduino-ide )

Have you tried just uploading?  I don't know what the "not connected" bit means, but it may be a Serial port (which won't be connected if you're in UF2 bootload mode) or OpenOCD debugging  (not yet supported).  The uplod script should copy the binary UF2 to whatever UF2 drive it finds.

earlephilhower, 2021-03-25:
Oops, the link where Les ran the 2.0 beta is https://www.tomshardware.com/news/raspberry-pi-pico-arduino-core-community .  But, same point, both 1.8 and 2,0-beta should be fine.

brandonplank, 2021-03-26:
Thanks, ill see.

brandonplank, 2021-03-26:
[image]
Honestly I'm surprised. I have no idea what's going on and i've tested on 4 different picos

brandonplank, 2021-03-26:
I've also tested on 1.8 with mild luck, its a hit or miss. 

On 1.8 i got it to send the uf2, but that was the end of that. It didn't work after that.

brandonplank, 2021-03-26:
[image]

brandonplank, 2021-03-26:
thats 1.8

earlephilhower, 2021-03-26:
Sorry, no idea.  You've got something odd going on that I don't think anyone else has hit.  Maybe someone else can chime in with suggestions.

A wild guess, as I'm not a big Windows or Windows Store user: maybe the Windows Store version of Arduino has some limited IO access (i.e. like an Android app with tight permissions) and maybe it can't see removable media?  I myself use the downloadable ZIP file on my Windows box without incident.  Can you try the normal EXE?

Also, are you running as a low-privileged user?  Vs. the usual Windows-ism of being a Local Admin?  That might restrict app access to  IO ports and drives, but it's just a guess.

brandonplank, 2021-03-26:
Ok, thanks for your help.
Figured out the issue!

You need to have the windows direct install for arduino 1.8 for the drivers!

Fixed all my problems!

closed 2021-03-26 by brandonplank

earlephilhower, 2021-03-26:
@brandonplank great news!

If I wanted to add something to the README about it, would this be correct?  If not, do you have a suggestion?

````
Windows Users:  Please do not use the Windows Store version of the actual Arduino application because it has issues detecting attached Pico boards.  Use the "Windows ZIP" or plain "Windows" executable download direct from https://arduino.cc. and allow it to install any device drivers it suggests.
````

brandonplank, 2021-03-26:
Yep! Looks good!

Would be great for users with this problem.

So to clarify.

To use even 2.0.0(beta), you need 1.8(Not from windows store, but from the website itself) to make sure the drivers are installed.

brandonplank, 2021-03-26:
Also as another update, It seems you need to have the 1.8 IDE open while you are working in 2.0.0.

If you do not, 2.0.0 does not detect the COM PORT.

fjansson, 2022-07-17:
I'm meeting what seems to be the same issue on Windows 10: a fresh install of the Arduino editor and arduino-pico cannot program a blank Pico or a Pico connected in BOOTSEL mode. 

This can be somewhat difficult to reproduce, because if you had it working once or Arduino at some point connected to a serial port, the com port setting is remembered in the Arduino preferences file and this makes subsequent Pico programming work.
Crucially, this issue will hit new users who install the Arduino editor to program a new Pico. 

## Steps
* Fresh install of Arduino 1.8.19 on Windows 10. Using the Windows .exe installer. Allow all drivers the installer proposes.
* Install arduino-pico according to instructions in this repo. I have version 2.3.0
* Select Tools -> Board -> Raspberry Pi RP2040 Boards(2.3.0) -> Raspberry Pi Pico
* Open Blink example
* Connect a Pico with USB in BOOTSEL mode (holding down BOOTSEL while plugging in the USB cable)
* observe Tools > Port menu is gray (at least for me it is, the computer has no other serial ports and Pico in BOOTSEL mode doesn't have a serial port)
* press Upload
  for me compilation works, upload fails with "Serial port not selected". I cannot select a serial port, because the Port menu is gray.

If you connect a Pico already programmed from Arduino in non-BOOTSEL mode, upload works. After this the Arduino editor remembers the serial port in the preferences file. Subsequent uploads also in BOOTSEL mode will work. One way I found to reproduce the problem is to delete Arduino's preferences file (when Arduino is not running). 
This PC > Windows (C:) > Users > <user name> > AppData > Local > Arduino15 > preferences
Then start Arduino, and fill in the arduino-pico Board Manager URL in File>preferences again. Now the port setting is forgotten, as you can verify in the status bar at the bottom of the window. Just reinstalling Arduino seems insufficient, as that doesn't delete the preferences file.

## Workaround

Program the Pico once by drag-and-drop a .uf2 file:
* Open a new, empty sketch (doesn't work with a read-only example sketch). 
* Compile it by pressing the "Verify" checkmark button. 
* Sketch > Export compiled Binary. Select a location e.g. the desktop. A folder is created there, containing a file ending in .uf2
* Copy this file to the Pico's drive, by drag and drop in the Explorer.
 
The Pico restarts and now now has a serial port. Now the Port menu in Arduino is not gray anymore, select the port there. After this, normal uploading from the Arduino editor should work.

ronie19, 2022-12-23:
My pi pico board unrecognised after uploading this sketch to communicate with SIM868 via Serial1 port pin 0 and 1. Bootsel mode and sketch upload is ok. After sketch upload USB fails to recognise by PC

What am I doing wrong here ?

char ponpin = 14;
int period = 1000;
unsigned long time_now = 0;
long previousMillis = 0;
long interval = 1000;
void(* resetFunc) (void) = 0;
//SoftwareSerial mySerial(7, 8);
String Tym, Loc, smsMode, gpsPwr, gpsSts, gpsRst, chg, chgen, cen, gpsmode, Loc1, gpswait, cn, tz, simrst;
String SMS_Msg;
float Latitude = 0.0, Longitude = 0.0;
const byte numChars = 100;
char receivedChars[numChars];
int count = 900;
void setup() {
pinMode (ponpin, OUTPUT);
digitalWrite(ponpin, HIGH);

Serial1.setRX(0);
Serial1.setTX(1);

  Serial1.setFIFOSize(128);
  Serial1.begin(9600);

  Serial.begin(9600);
  simrst = gsmQuery("AT+CFUN=1,1");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(simrst);
  chgen = gsmQuery("AT+ECHARGE=1");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(chgen);
  tz = gsmQuery("AT+CLTS=0");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(tz);
  cn = gsmQuery("AT+CNMI=0,0,0,0,0");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(cn);
  gpsPwr = gsmQuery("AT+CGNSPWR=1");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(gpsPwr);
  chg = gsmQuery("AT+CBC");
  //Serial.println(chg);
  gpsRst = gsmQuery("AT+CGNSSEQ=RMC");
  time_now = millis();
  while (millis() < time_now + period) {}
  Serial.println(gpsRst);
  pinMode(LED_BUILTIN, OUTPUT);
  //gpswait = gsmQuery("WAIT=15");
  Serial.println(gpswait);
}

void loop() {
  
}

String gsmQuery(String query) {
  String f;
  f = "";
  Serial1.print(query);
  Serial1.print("\r\n");
  while (Serial1.available() == 0)  {}
  f = Serial1.readString();
  while (Serial1.available() > 0)  {}
  query = "";
  return f;
}

String getValue(String data, char separator, int index)
{
  int found = 0;
  int strIndex[] = {0, -1};
  int maxIndex = data.length() - 1;

  for (int i = 0; i <= maxIndex && found <= index; i++) {
    if (data.charAt(i) == separator || i == maxIndex) {
      found++;
      strIndex[0] = strIndex[1] + 1;
      strIndex[1] = (i == maxIndex) ? i + 1 : i;
    }
  }

  return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
}

String GetStringBetweenStrings(String input, String firstdel, String enddel) {
  int posfrom = input.indexOf(firstdel) + firstdel.length();
  int posto   = input.indexOf(enddel);
  return input.substring(posfrom, posto);
}