earlephilhower/arduino-pico #21: Compile Error for attachinterrupt()

aayushgavande, 2021-03-26:
I assume the attachInterrupt() function has not yet been added, it would be great if you add it, as interrupts using micro python are really slow and thanks a lot for this repo.

earlephilhower, 2021-03-26:
@aayushgavande are you able to test against the GIT version of the tree?  Adding `attachInterrupt` supports looks pretty simple thanks to the great Pico SDK, but I don't have any really good test cases other than shorting a pin with a jumper...

aayushgavande, 2021-03-26:
I will try installing it with the GIT version, and talking about the test cases you can always try using different sensors with it or you can use a frequency generator to create the pulse with desired frequency and then doing some operations in the isr like incrementing the count. 

Thank You.

earlephilhower, 2021-03-26:
See #22.  In the worst case you can just overwrite the two changed files in your installation and give it a go.

aayushgavande, 2021-03-26:
> See #22. In the worst case you can just overwrite the two changed files in your installation and give it a go.

Thanks a lot works great.

PR #22 (2021-03-26): Add attach/detachInterrupt support
Fixes #21
commit 09b5693f (2021-03-26): Add attach/detachInterrupt support

Fixes #21
closed 2021-03-26 by aayushgavande