earlephilhower/arduino-pico #23: Building with arduino-cli

joachimhs, 2021-03-26:
Hello, 

Thank you for taking the time to create this library! I am not sure if this is a bug or not, but I am attempting to build the standard blink sketch using Arduino-cli. I have added the board to the YAML configuration file, and inside Arduino IDE I can compile and upload the sketch just fine. 

However, using the latest version of Arduino-cli I am getting the following error, which I am not sure how to fix. 

`joachimhaagenskeie@Joachims-MacBook-Pro b989c882d2b % arduino-cli compile -e -b rp2040:rp2040:rpipico /Users/joachimhaagenskeie/src/arduinocli/b989c882d2b/b989c882d2b.ino
/Users/joachimhaagenskeie/Library/Arduino15/packages/rp2040/hardware/rp2040/0.9.5/cores/rp2040/main.cpp: In function 'int main()':
/Users/joachimhaagenskeie/Library/Arduino15/packages/rp2040/hardware/rp2040/0.9.5/cores/rp2040/main.cpp:27:5: error: 'DEBUG_RP2040_PORT' was not declared in this scope
   27 |     DEBUG_RP2040_PORT.begin();
      |     ^~~~~~~~~~~~~~~~~
/Users/joachimhaagenskeie/Library/Arduino15/packages/rp2040/hardware/rp2040/0.9.5/cores/rp2040/main.cpp: In function 'ssize_t _write(int, const void*, size_t)':
/Users/joachimhaagenskeie/Library/Arduino15/packages/rp2040/hardware/rp2040/0.9.5/cores/rp2040/main.cpp:62:12: error: 'DEBUG_RP2040_PORT' was not declared in this scope
   62 |     return DEBUG_RP2040_PORT.write((const char *)buf, count);
      |            ^~~~~~~~~~~~~~~~~

Error during build: exit status 1`

earlephilhower, 2021-03-26:
You're missing a define from the IDE.  See `boards.txt` for the list, but a quick fix would be to add `-DDEBUG_RP2040_PORT=Serial` to the build.

closed 2021-03-26 by earlephilhower

joachimhs, 2021-03-26:
How would I go about to add this to the arduino-cli build. It doesn't seem to have this option. I attempted to add a define at the top of my sketch, but that didn't work. I could, however add this define to the main.cpp file inside the rp2040 core. 

`#define DEBUG_RP2040_PORT Serial`

This will enable me to compile the project to a UF2 file, however this change will disappear if I choose to update the core in the future.

earlephilhower, 2021-03-26:
What you've done is fine, and I've already taken care of this case in the current `master` so the next release 0.9.6 will work as-is, no changes needed, in arduino-cli.  Check commit https://github.com/earlephilhower/arduino-pico/commit/1926473f5bdb91d6e09aeee67a7d0c3bde6a8f3a

joachimhs, 2021-03-26:
Ah, thank you for your help!

commit 1926473f (2021-03-26): Avoid compile error when debug port unspecified

Fixes #23