earlephilhower/arduino-pico #25: Allowing Multiple interrupts

aayushgavande, 2021-03-26:
can you please come up with an update for taking more than one interrupts on the same core.

earlephilhower, 2021-03-26:
Sorry, I don't understand what you're requesting.

You should be able to have an IRQ per pin now.  However, multiple IRQs on the same pin is not supported by Arduino (HW limitations), but there are workarounds 9(i.e. use CHANGE and in the IRQ handler see if gpio is 1 or 0 to figure out if it was RISING or FALLING).

aayushgavande, 2021-03-26:
I have two different isr() for two different interrupts which i am taking on two different pins at a different time the issue I am facing is that only one isr is getting called even if interrupt takes place for different pin. 

ex. 

void isr1()
{
  //do something
}

void isr2()
{
//do something 
}

void setup()
{
   //set pinmode to input 
   attachInterrupt(pin1 , isr1 , RISING);
   attachInterrupt(pin2 , isr2 , FALLING);

}

void loop()
{

}

in this case, if there is an interrupt at pin2 isr2 should be called and if there is an interrupt at pin1 isr1 should be called but for some reason, isr2 is getting called for both cases. This is the problem i am facing.

earlephilhower, 2021-03-26:
Ah, I see.  I think my understanding of the RPI GPIO interrupt is mistaken, then.  I assumed each pin had its own handler but the code actually has this:
````
void gpio_set_irq_enabled_with_callback(uint gpio, uint32_t events, bool enabled, gpio_irq_callback_t callback) {
    gpio_set_irq_enabled(gpio, events, enabled);

    // TODO: Do we want to support a callback per GPIO pin?
    // Install IRQ handler
    _callbacks[get_core_num()] = callback;
    irq_set_exclusive_handler(IO_IRQ_BANK0, gpio_irq_handler);
    irq_set_enabled(IO_IRQ_BANK0, true);
}
````

Looks like TODO bit us on the bum!

earlephilhower, 2021-03-26:
@aayushgavande please give #26 a try and report back.

aayushgavande, 2021-03-26:
> @aayushgavande please give #26 a try and report back.

I just gave it a shot and it works great. Thanks a lot for such a quick response.

PR #26 (2021-03-26): Fix attachInterrupt to handle single GPIO callback
The Pico SDK only supports a single callback pointer for all GPIO
interrupt pins.  So we need to track the individual pin-to-CB map
ourselves and dispatch from our own internal callback routine.

Fixes #25
commit b96c3716 (2021-03-26): Fix attachInterrupt to handle single GPIO callback

The Pico SDK only supports a single callback pointer for all GPIO
interrupt pins.  So we need to track the individual pin-to-CB map
ourselves and dispatch from our own internal callback routine.

Fixes #25
closed 2021-03-26 by aayushgavande