earlephilhower/arduino-pico #28: No /media mount point

ccpetersen, 2021-03-26:
It would appear the mount point for the pico is hardcoded to be "/media".  In Manjaro(Arch) usb mount points are in "/run/media". The simple solution is to create a link "/run/media -> /media", works but should be fixed.

vslinuxdotnet, 2021-03-26:
Yes it hard coded in  uf2conv.py in line: 238 
rootpath = "/media"

Should be changed to something like this:

```
rootpath = "/media"
if not os.path.isdir(rootpath):
   rootpath = "/run"
```

earlephilhower, 2021-03-27:
I'll manually add this here, but I think you want to also open an issue/PR upstream where the file comes from:  https://github.com/microsoft/uf2

earlephilhower, 2021-03-27:
@ccpetersen Can you check out #37 and report back?  Thx!

PR #37 (2021-05-07): Look for UF2 drives in additional spots on Linux
Fixes #28
commit f1e77995 (2021-05-07): Look for UF2 drives in additional spots on Linux (#37)

Fixes #28
closed 2021-05-07 by earlephilhower