earlephilhower/arduino-pico #29: List of Pico-Specific Functions?

DanielDC88, 2021-03-26:
This may be a silly question, since I am a novice with both arduino and pico. I can't seem to find the documentation on the functions and syntax used to control the pins on the pico.

Please could you direct me to the right place?

marcelvanherk, 2021-03-26:
Hi, just use the standard Arduino function, pinMode, digitalWrite and digitalRead.

earlephilhower, 2021-03-26:
@DanielDC88 it's as boring standard Arduino as I can make it.  https://arduino.cc is your best bet, along with hundreds of online tutorials.

Note that this is an unofficial port, written by one guy over a 3-day weekend, and still in 0.9.x quality, so things may be broken.

closed 2021-03-26 by earlephilhower

DanielDC88, 2021-03-27:
@earlephilhower thanks for  the reply. I wasn't sure of what the pins would be, but I think I've been able to work this out since. The functions pinMode, digitalWrite and digitalRead all seem to work.

Is something equivalent to this implemented? https://www.arduino.cc/en/Reference/PortManipulation
I have tried these commands and they don't get recognised at the moment.

earlephilhower, 2021-03-27:
@DanielDC88, the `PORT...` registers are actual hardware registers on the original 8-bit AVR and not supported on anything else.  Use the `pinMode` and `digitalWrite`/etc. and you will be fine.