earlephilhower/arduino-pico #32: Integer / Floats undefined behavior

brandonplank, 2021-03-26:
[image]
yeah, the screenshot speaks for itself.

earlephilhower, 2021-03-26:
Your format string is using '%d %d %d` which means you must pass in an `int` not a `float`.  You might even get a compile warning about it...

You'll want `%f` for floats...

brandonplank, 2021-03-26:
tried %f. Its does not even output the float.
It outputs nothing.

brandonplank, 2021-03-26:
That why i used %d. and i even casted it to a int using (int)

brandonplank, 2021-03-26:
[image]

brandonplank, 2021-03-26:
See. Its odd

earlephilhower, 2021-03-26:
Ah, that is interesting and proper code.  The newlib printf may be linked w/o floating point support.  Let me do a little digging, or you can check out the libc.a yourself in parallel...

earlephilhower, 2021-03-26:
Oh, and in the meantime `Serial.print(<float>)` does work (it uses `dtostrf`, a separate function provided by the core).

earlephilhower, 2021-03-26:
Trivial linking fix.  0.9.7 should get you going.

commit c50f1ded (2021-03-26): Fix printf/scanf of float and doubles

Needed to fix link stage to pull in `%f` formatting support.

Fixes #32
closed 2021-03-26 by earlephilhower