earlephilhower/arduino-pico #35: change clock don't work

vslinuxdotnet, 2021-03-27:
Hello,
using the function set_sys_clock_khz() without the include of "pico/stdlib.h", report missing function...
But the include is set in platform.txt: "-I{runtime.platform.path}/pico-sdk/src/common/pico_stdlib/include"
if I add the include "pico/stdlib.h", I get multiples errors in linking: multiple definition of `_write' && multiple definition of `_read';
As I can fix that?

Thanks

earlephilhower, 2021-03-27:
Latest libpico.a should work.  Just `#iclude <pico/stdlib.h>` and you can call the fcn.  Issue was related to having the Pico SDK's stdio-usb hack stuff still (unused) in the lib.a file.

closed 2021-03-27 by earlephilhower

SensorsIot, 2021-03-28:
Hello Erle,
set_sys_clock_khz(290400, true); works now. Unfortunately, the USB is no more recognized on my Windows PC after this command. Only if I flash the script without this command (using the bootsel method) USB works again. Is this to be expected?

'vreg_set_voltage' was not declared in this scope.

earlephilhower, 2021-03-28:
You can't change the clock once things are running because the whole machine is based off of it, and the USB interface will then be configured to run at the wrong speed.

Check the current master.  I just added a menu to overclock and got 250MHz stable with simple tests.

SensorsIot, 2021-03-28:
Cool. Thank you. Very simple solution. Mine also works up to 250 MHz. It seems that if we want to go higher we would need to increase the core voltage (vreg_set_voltage(VREG_VOLTAGE_1_30) ).

vslinuxdotnet, 2021-03-28:
Confirm that seams the works fine up to 250 Mhz, and Serial/USB/Timings keeps synchronized.
I also checked the temperature of the CPU using the internal sensor, increased +/- more 2ºC to run at 250mhz.

PR #36 (2021-03-27): Remove stdio hacks from libpico.a
Fixes #35
commit 90ba9424 (2021-03-27): Remove stdio hacks from libpico.a

Fixes #35