earlephilhower/arduino-pico #38: zero-byte writes in arduino Wire must be handled by bitbanging

ladyada, 2021-03-27:
a common thing done in arduino libraries is performing a zero-byte-write, like this
https://github.com/arduino-libraries/ArduinoECCX08/blob/master/src/ECCX08.cpp#L440
or in address scanning/checking

the RP2040 I2C peripheral does not permit that, so for writes of 0 bytes, it must be bitbanged, we've done so in CircuitPython and MicroPython does the same

right now, because of this bug, a lot of libraries (including a lot of Adafruit ones) will not work, they will fail on the address detection step!

(cc @sandeepmistry in case Arduino's core needs it too)

ladyada, 2021-03-27:
(besides that, this core works really well! good job :)

brandonplank, 2021-03-28:
Thumbs up to this!

earlephilhower, 2021-03-28:
Thanks, @ladyada and @brandonplank!

I don't really have many I2C devices to test on, but I've run it against the other I2C port on the Pico and patch #40 seems to work properly with special-cased 0-len I2C reads.  (I ran this I2S scanner: https://gist.github.com/tfeldmann/5411375 and a few ad-hoc tests).

Would one of you be able to test on your own setup and report back, please?

brandonplank, 2021-03-28:
Yep! Let me get everything hooked up on my end and i'll let you know!

ladyada, 2021-03-28:
tested, the sensors i tried before are now working :)

earlephilhower, 2021-03-28:
Thanks for the update @ladyada !

PR #40 (2021-03-28): Fix INPUT_PULLUP/DOWN, I2C 0-len probes, timeout
Fix the drive logic for pullup and pulldown inputs.

Add a bitbanged I2C ACK checker for 0-byte I2C transfers (since the HW
doesn't support it).

Add a timeout on I2C reads and writes so they don't hang forever if a
device is not present.

Fixes #38
commit 4076bf85 (2021-03-28): Fix INPUT_PULLUP/DOWN, I2C 0-len probes, timeout

Fix the drive logic for pullup and pulldown inputs.

Add a bitbanged I2C ACK checker for 0-byte I2C transfers (since the HW
doesn't support it).

Add a timeout on I2C reads and writes so they don't hang forever if a
device is not present.

Fixes #38
closed 2021-03-28 by earlephilhower