earlephilhower/arduino-pico #41: Won't Install on Raspberry Pi

JohnInWales, 2021-03-28:
Trying to install using Board Manager (Arduino IDE 1.8.13) on a Raspberry Pi 4 4Gb using Raspberry Pi OS 32-bit, and get error "Tool pqt-gcc is not available for your operating system."

It installed, and so far works, with no problems in Linux Mint on another machine.

yoramrotbach, 2021-03-28:
I confirm same behavior

Yoram

TestUserOnlineNL, 2021-03-28:
Please fix this problem so that we can do development on the raspberry pi. I got the same error. Many thanks in advance.

arijus303, 2021-03-28:
RPI3  B+ , doesn't install either , im getting : "Tool pqt-elf2uf2 is not available for your operating system"  

Running on Raspbian GNU/Linux 10 (buster)
original C/Cmake tool-chain works and compiles fine.
Any help would be great, thx in advance

earlephilhower, 2021-03-28:
Thanks, this is a known issue.

I compile all tools from scratch on my Linux x86-64 server:  Mac, Windows, AARCH64, etc.  To do this, I need cross-compilers for the targets.  I don't even have a Mac or an ARM64 system, but I support them for this and the ESP8266 using this setup.

Unfortunately the Raspberry PI cross compiler available from the RPI project is ancient 4.xxx and can't build the pico-sdk because it doesn't support all the C++ features it uses.

If you know of a newer cross-compiler, please do let me know and I can add it back to the pico-quick-toolchain build.

Making the RPI binaries is trivial w/the right x-compiler...literally remove one `#` and redo `make -j32` and sit back for 30 minutes...

JohnInWales, 2021-03-28:
That's a shame. It's a computer it should work on, as it comes from the same manufacturer, and Raspberry Pi encourage people to use them to program Picos! But if it's not possible I can live with it.

If you haven't already, and I've missed it, could you put a note in the read me file to explain this, as I'm sure a lot of people will get caught out by it?

earlephilhower, 2021-03-28:
@JohnInWales sorry, good point.  I **do** want to add RPI support, but need someone to point me to a proper cross-compiler.   I'll update the readme with a little note.

earlephilhower, 2021-03-28:
Good news, @JohnInWales @arijus303 @TestUserOnlineNL @yoramrotbach .  It looks like Ubuntu has a gcc-7 cross-compiler available as part of the standard repos now!

I will attempt to rebuild the Docker build system and then re-add the RPI toolchain builds and leave a note here for you to test them out later today...

earlephilhower, 2021-03-28:
@JohnInWales @arijus303 @TestUserOnlineNL @yoramrotbach - Can one of you please give release 0.9.11 a try on a real Raspberry Pi and report back?

JohnInWales, 2021-03-28:
Thanks for finding a solution so fast. It installed without any problem, but fails when compiling your "Fade" example. I've pasted the full log below, but the actual error is:

/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3: no such file or directory

I've updated the Mint machine, and that compiles with no problems.

On the Pi, the "1.0.1-base-3a57aed" directory contains the directories "bin" and "share", and file "package.json"
On my Linux Mint PC, the directory contains "placeholder_for_arduino" (an empty file), and "python3"

Let me know if you need more information.

----------------------------------------------------------------------------------------------------------------

Arduino: 1.8.13 (Linux), Board: "Raspberry Pi Pico, 2MB (no FS), 125 MHz, Disabled, None"

/opt/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/pi/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/pi/Arduino/libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=0000_0000 -ide-version=10813 -build-path /tmp/arduino_build_682589 -warnings=none -build-cache /tmp/arduino_cache_604593 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-python3.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-openocd/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-openocd/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-pioasm/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-pioasm/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.1.0-a-81a1771 -verbose /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11/libraries/rp2040/examples/Fade/Fade.ino
/opt/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /opt/arduino-1.8.13/hardware -hardware /home/pi/.arduino15/packages -tools /opt/arduino-1.8.13/tools-builder -tools /opt/arduino-1.8.13/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /opt/arduino-1.8.13/libraries -libraries /home/pi/Arduino/libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=0000_0000 -ide-version=10813 -build-path /tmp/arduino_build_682589 -warnings=none -build-cache /tmp/arduino_cache_604593 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-python3.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-openocd/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-openocd/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-pioasm/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-pioasm/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=/home/pi/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.1.0-a-81a1771 -verbose /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11/libraries/rp2040/examples/Fade/Fade.ino
Using board 'rpipico' from platform in folder: /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11
Using core 'rp2040' from platform in folder: /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11
Detecting libraries used...
/home/pi/.arduino15/packages/rp2040/tools/pqt-gcc/1.1.0-a-81a1771/bin/arm-none-eabi-g++ -c -Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -I/home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11/pico_b [... cut ...]
/home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11/tools/simplesub.py --input /home/pi/.arduino15/packages/rp2040/hardware/rp2040/0.9.11/lib/memmap_default.ld --out /tmp/arduino_build_682589/memmap_default.ld --sub __FLASH_LENGTH__ 2093056 --sub __EEPROM_START__ 2093056 --sub __FS_START__ 2093056 --sub __FS_END__ 2093056
fork/exec /home/pi/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3: no such file or directory
Error compiling for board Raspberry Pi Pico.

earlephilhower, 2021-03-28:
@JohnInWales Thanks.  Easy fix.  I just had the python link wrong for the RPI.  0.9.12 just put up should correct it.

JohnInWales, 2021-03-29:
Thanks. Successfully uploaded Fade, and it works. I'll try something more adventurous later.

earlephilhower, 2021-03-29:
Thanks for the update!  Glad it's working.

closed 2021-03-29 by earlephilhower

TestUserOnlineNL, 2021-03-30:
Many thanks for fixing the issue. Great work. 🤗👍

PR #45 (2021-03-28): Add Raspberry Pi binaries, OpenOCD, and new boards
Fixes #41