earlephilhower/arduino-pico #42: Please support Adafruit Feather RP2040 with 8MB flash

yoramrotbach, 2021-03-28:
Hi,

I would appreciate if you'll support Adafruit Feather RP2040 which has 8MB flash.
I've found in the code where it says that the pico has only 2048k flash, but I don't know if changing this to 8096k will make it work.

Thanks
Yoram

earlephilhower, 2021-03-29:
So the PR did not actually fix this.

It turns out that the Pico has **2 different and incompatible** ROM versions.  The ones on the Raspberry Pi foundation devices, like the Pico, are B0 and Adafruit and others are running B1.

Normally when you rev a ROM the interfaces stay the same, but in this case they don't so you actually need **different code** to run on B0 vs. B1, and I think that's the issue.

I think just rebuilding libpico.a with a proper config option will help here.

earlephilhower, 2021-03-29:
@yoramrotbach  Please give #42 a try and report back.

yoramrotbach, 2021-03-29:
@earlephilhower, should there be a new version available for update? I'm running 0.9.12

earlephilhower, 2021-03-29:
@yoramrotbach are you sure your board is not busted in some way?

Watch this video where ladyada at Adafruit shows a demo running the same board you have, without any changes to the core (i.e. this was just after I fixed the I2C 0-byte probe stuff). - https://www.youtube.com/watch?v=fKDeqZiIwHg

She also runs a QT Py RP2040 pre-release and it runs fine. too.

So I'm inclined to think this may be something specific to your setup and not the core.  As I don't have anything but the official Picos on-hand I can't really do any testing.  If there are other users w/the Feather they can chime in, too.

yoramrotbach, 2021-03-30:
@earlephilhower, I don't know how it could be busted if it working perfectly with CircuitPython. I saw ladyada videos and it is strange that she got it running, however she doesn't explain what she did or if it worked out of the box.
I've followed the changes you did to the code and was thinking that on the IDE boards manager there will be a new version to download like 0.9.13 that will contain the B1 changes you did. currently I don't have the new files to test with. If I should built it on my own based on your changes, just let me know.
Thank you very much for your support

Yoram

earlephilhower, 2021-03-30:
I didn't make a new release w/the change because it's not been validated yet.

Why not get the 0.9.13 release, just put out, and then
* Download this file: https://github.com/earlephilhower/arduino-pico/raw/302ca03e6ac9bf29891fda8ad2361350526d1ba9/lib/libpico_b1.a
* Rename it to `libpico.a`
* Copy it over the `libpico.a` file that's part of the release
* Rebuild and upload Fade.ino

If that doesn't help, then your issue is not related to the ROM version and the PR should be dropped, anyway, as not useful...

earlephilhower, 2021-03-30:
Ah...looks like the pico-sdk may have issues w/your board!  I was just checking on a PR I had there and saw this:  https://github.com/raspberrypi/pico-sdk/pull/279

The clock divider 2 vs 4 would cause flash to just not work (it'd be too fast at /2 which is what is in the SDK).  There's also something going on w/the boot2 stage, but I don't understand his comments there.

Guess that's what we get for bring on the bleeding edge!

yoramrotbach, 2021-03-30:
Thanks Earl for your support.
It does look like there are serious compatibility issues between the SDK and the feather flash chip.
Adafruit have solved it for their own CircuitPython.
What I don't get is how ladyada is showing your code running on the QT py. probably a different flash chip which the SDK supports.
I will continue to monitor this topic and hopefully the SDK team will have better support for various flash chips soon.
I'm looking forward to developing in Arduino IDE on the RP2040

blurfl, 2021-03-30:
> ... get the 0.9.13 release, just put out, and then
>
> * Download this file: https://github.com/earlephilhower/arduino-pico/raw/302ca03e6ac9bf29891fda8ad2361350526d1ba9/lib/libpico_b1.a
> * Rename it to [...]
I've done these steps, and still find that the 'Adafruit Feather RP2040' definition has issues. Compiling and loading the Bootsel example completes as expected (uf2 method), but after rebooting the board shows no serial device. I note that  the new libpico.a is smaller than the one it replaces, is that expected?

earlephilhower, 2021-03-30:
@blurfl see https://github.com/earlephilhower/arduino-pico/issues/42#issuecomment-810297266 .  There are some issues w/the pico-sdk currently.  Adafruit has patched their fork, but the upstream (which is what I'm using) has not yet been updated.

aallan, 2021-03-31:
> What I don't get is how ladyada is showing your code running on the QT py. probably a different flash chip which the SDK supports.

The SDK supports the Flash chip on the Feather perfectly fine, but you have to compile your code using the generic03 (slow!) flash profile rather than the default winbond profile (used on Pico), see raspberrypi/pico-sdk#275.

earlephilhower, 2021-03-31:
@aallan Thanks very much for the pointer!  I've been able to build and ship all 8 possible boot sectors (/2 and /4 for each of the 4) with the pointer where to look you gave.  I'm a maintainer on the ESP8266 Arduino core, and we also have issues with different/incompatible flash chips....

@yoramrotbach @blurfl if you can use GIT, can you try #67 out?  I have put the Feather at the slowest, most compatible setting (potato SPI/4).  That should come up no matter what.  If so, then we can figure out the "right" one that's not a potato (aka 3sx faster flash read)

Then select the Generic board, select 8MB, and the go through the 8 possible options in the new Boot Stage 2.  I think the "w25q080 /4" is equivalent to what Adafruit is running (but they actually wrote their own custom boot stage 2 so it's not *exactly* the same), but there is no harm in trying everything and seeing what works.

aallan, 2021-03-31:
> @aallan Thanks very much for the pointer! 

No worries. This wasn't an issue we were really thinking about until the Feather shipped.

earlephilhower, 2021-03-31:
Try release 0.9.15, just put up.  I merged after more testing as it didn't seem to break anything on the genuine Pico side.

blurfl, 2021-03-31:
> can you try #67 out?

Release 0.9.15 seems to work correctly with the Feather in the example sketches 'Fade', 'Temperature' and 'Bootsel' 👍

earlephilhower, 2021-03-31:
Awesome, @blurfl .  Can you try this experiment and see which bs2 works best for you?  Right now you are running something that is much slower (but more compatible) than other boards when reading the flash.

> Then select the Generic board, select 8MB, and the go through the 8 possible options in the new Boot Stage 2. I think the "w25q080 /4" is equivalent to what Adafruit is running (but they actually [...]

earlephilhower, 2021-03-31:
Any of them.  It reads the flash to run your code, and there's a lot of code in just the USB serial port.  I used BOOTSEL which is small and fast.  Change the boot stage, reflash, see if the serial port still lives and if it reacts to button presses; then repeat.

blurfl, 2021-03-31:
Here's what I see using 'bootsel.ino':
Generic SPI /2 - works as expected
Generic SPI /4 - works as expected
IS25LP080 QSPI /2 - loses serial port after uf2 load, doesn't seeem to run
IS25LP080 QSPI /4 - loses serial port after uf2 load, doesn't seeem to run
W25Q080 QSPI /2 - loses serial port after uf2 load, doesn't seeem to run
W25Q080 QSPI /4 - loses serial port after uf2 load, doesn't seeem to run
W25X10CL QSPI /2 - flags this error compiling:
.../assembly/boot2_w25x10_2_padded_checksum.S: No such file or directory
 I found that renaming boot2_w25x10cl_2_padded_checksum.S allows the compile to complete. The serial port is available after the uf2 upload and the program runs then, but after rebooting the Feather, it loses serial port after uf2 load, doesn't seem to run
W25X10CL QSPI /4 - after renaming ...checksum.S file, works as expected.

earlephilhower, 2021-03-31:
Thanks for the feedback.  That `W25X10CL QSPI /4` is significantly faster than what you started with.  I've updated the name and Adafruit Feather RP2040 defaults, so the next release you should be all set (no need to use generic/etc.)

closed 2021-03-31 by earlephilhower

yoramrotbach, 2021-04-01:
Thanks @earlephilhower 
Your message came during my night, so I've tested it now and can also confirm that it is working.
Thanks again for not giving up and providing an amazing support.

Yoram

PR #44 (2021-03-28): Adafruit Feather, overclocking, flash FS shims
Add support for Adafruit Feather RP2040 (8MB).  Identified with a unique
USB PID so it displays properly once programmed one time.  Moved LED to
pin 13 (per the website docs), but have no board to test.

Add over/underclocking menus, applied at boot.  Use at your own risk, as
usual.

Add shims to allocate flash space for a filesystem (but not implemented
yet).

Add a "generic" RP2040 board

Fixes #42
PR #67 (2021-03-31): Add multiple boot2 options with /2 and /4 SPI div
Build all 4 different boot_stage2 objects with SPI/2 and SPI/4 options.
Add a menu to the generic board to allow selecting between them.
Use the potato/4 boot2 for the Adafruit Feather until we have a better
version verified.

Fixes #42
commit 09268fe0 (2021-03-31): Add multiple boot2 options with /2 and /4 SPI div (#67)

Build all 4 different boot_stage2 objects with SPI/2 and SPI/4 options.
Add a menu to the generic board to allow selecting between them.
Use the potato/4 boot2 for the Adafruit Feather until we have a better
version verified.

Fixes #42