earlephilhower/arduino-pico #50: Opening a console window on the port does not reset

JohnGaby, 2021-03-29:
If I open a console window on the serial port for the pico, it does not reset the pico like it does for Arduino boards. Is this the expected behavior? Is there any way to make it auto reset when the port is opened?

Also, you might note in your readme file that if the first sketch you upload does NOT contain a Serial.begin call, the pico port will not show up.

Thanks.

blurfl, 2021-03-29:
`the first sketch you upload does NOT contain a Serial.begin call, the pico port will not show up.`

It seems here like uploading a sketch that doesn't contain Serial.begin() removes the pico serial port. the port only seems to appear when the most recent sketch called Serial.begin(). That may be intentional, I don't know?

earlephilhower, 2021-03-29:
> If I open a console window on the serial port for the pico, it does not reset the pico like it does for Arduino boards. Is this the expected behavior? Is there any way to make it auto reset when the port is opened?

Opening not resetting...that is expected behavior.  The USB is SW driven and if you reset it takes several seconds to boot the Pico far enough and negotiate things...and you've lost the serial port you were connected to anyway since it disappeared when reset.  The USB serial will reset when connected at 1200bps, like other USB-based Arduino boards.

> Also, you might note in your readme file that if the first sketch you upload does NOT contain a Serial.begin call, the pico port will not show up.

You found a bug.  It is started automatically, **assuming you enabled a debug port**.  But, if you don't manually set it at least once, Arduino IDE will not select one for you and it will not start up.  I'll fix this in an upcoming PR.  Thx!

PR #51 (2021-03-29): Always enable USB serial port for reset
To allow for uploads when sketches don't manually start the Serial port,
start the USB port always in main().

Fixes #50
closed 2021-03-29 by earlephilhower