earlephilhower/arduino-pico #52: When this BSP can support I2S?

nerodu, 2021-03-29:
Hi, 
I want to use pico and mic, does this support I2S now? Or any plan in the future?
Thanks

vslinuxdotnet, 2021-03-29:
The RPI Pico don't have DAC or I2C dedicated pins, so as implemented the signals will not be great quality.
You can see this project: https://github.com/rgrosset/pico-pwm-audio

earlephilhower, 2021-03-29:
Actually, @vslinuxdotnet , the PIO machines can do I2S, too.  I use them to do Servo control and Tone generation already.  So, it could have up to 8, theoretically. :)

In pico-extras there's a pre-final version of an I2S device and driver.  I haven't had time to look at it, but if someone else has then a PR would be most welcome...

nerodu, 2021-03-30:
Thanks for help. I also find the I2S in pico extras and I will try to integrate it .

closed 2021-03-30 by nerodu

stolk, 2022-04-11:
The I2S, as added to arduino-pico, only allows for I2S writes.
I2S reading, like OP wants to do for microphone, is not supported: all `read()` operations return `-1`.