earlephilhower/arduino-pico #57: Upload of a substantial program crashes the Arduino IDE

technoblogy, 2021-03-30:
I've now successfully uploaded small programs, such as Blink, and a Serial test. But when I upload a more substantial program of about 80Kbytes when compiled (my uLisp Lisp interpreter) it seems to crash the Arduino IDE on my Mac in a way that even a Force-Quit doesn't fix, and I have to restart my Mac to continue. So this is making further progress a bit slow.

Any suggestions?

earlephilhower, 2021-03-30:
@technoblogy the upload stage is pretty simple and uses the standard UF2 upload tool from Microsoft.  How that would cause your Mac to freeze, I have no idea unfortunately.  I'd check physical things, especially USB cables, as they sometimes are underspec'd and give power droops under extended load.

FWIW, large projects have been done (see https://github.com/earlephilhower/arduino-pico/issues/11 ) without any specific upload issues.

Anyone on a Mac, please feel free to chime in!

You could also use a PicoProbe (see 0.9.13) which would use a different upload method buy selecting a different board in the menus.

technoblogy, 2021-03-30:
OK, thanks. I'll simplify my program and then reintroduce functionality in stages to see at what point it breaks.

marcelvanherk, 2021-03-30:
I compile on windows, and indeed my code is fairly big 731k UF2; no problems

blurfl, 2021-04-06:
Testing with @marcelvanherk 's code on a Mac with Arduino 1.8.14-hourly, the 726K UF2 uploads without error. FWIW I've had other issues with the Arduino IDE on macos 11.3 but they are unrelated to any individual project.

technoblogy, 2021-04-06:
Thanks - the problem must be with my code.

earlephilhower, 2021-04-08:
Given the discussion, I'm going to close this for now.

closed 2021-04-08 by earlephilhower