earlephilhower/arduino-pico #58: Adafruit OLED/TFT libraries don't compile

ProxyPlayerHD, 2021-03-30:
[Adafruit_SSD1306](https://github.com/adafruit/Adafruit_SSD1306) and Adafruit_ST7735 (the included TFT library) to be exact.

hi, I'm trying to experiment around with some displays on the Pico, but i'm just hitting a brickwall. here is the code for both programs: [OLED](https://pastebin.com/cwNBUzD9), [TFT](https://pastebin.com/MCWwj7P3)

using the Adafruit_SSD1306 Library i'm getting this error:

```
C:\Users\x\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp: In member function 'bool Adafruit_SSD1306::begin(uint8_t, uint8_t, bool, bool)':
C:\Users\x\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:495:44: error: 'digitalPinToPort' was not declared in this scope; did you mean 'digitalPinToInterrupt'?
  495 |     dcPort = (PortReg *)portOutputRegister(digitalPinToPort(dcPin));
      |                                            ^~~~~~~~~~~~~~~~
      |                                            digitalPinToInterrupt
C:\Users\x\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:495:25: error: 'portOutputRegister' was not declared in this scope
  495 |     dcPort = (PortReg *)portOutputRegister(digitalPinToPort(dcPin));
      |                         ^~~~~~~~~~~~~~~~~~
C:\Users\x\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:496:17: error: 'digitalPinToBitMask' was not declared in this scope
  496 |     dcPinMask = digitalPinToBitMask(dcPin);
      |                 ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Raspberry Pi Pico.
```

with the other Library it was a bit more complicated.

at first i was getting this error:
```
In file included from C:\Users\oskar\Documents\Arduino\libraries\TFT\src/TFT.h:37,
                 from C:\Users\oskar\Documents\Arduino\Programs\TFT_Testing\TFT_Testing.ino:1:
C:\Users\x\Documents\Arduino\libraries\TFT\src/utility/Adafruit_ST7735.h:30:10: fatal error: avr\pgmspace.h: No such file or directory
   30 | #include <avr\pgmspace.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Raspberry Pi Pico.
```
after a bit of googling i found some dummy header files in an AVR folder that i was able to just throw into `cores\RP2040` and that seem to have worked. Link to [SO thread](https://stackoverflow.com/questions/66814408/e-paper-display-and-raspberry-pi-pico)
so the next error i got after that was:
```
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\glcdfont.c:3:10: fatal error: avr/io.h: No such file or directory
    3 | #include <avr/io.h>
      |          ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Raspberry Pi Pico.
```
which was caused by this piece of code:
```
#if !defined(ARDUINO_ARCH_SAM) && !defined(__ARDUINO_ARC__) && \
    !defined(ARDUINO_ARCH_STM32)
#include <avr/io.h>
#endif
```
it seems to only include `avr/io.h` if the MCU used is not a 32 bit ARM one... so i just commented the whole thing out and hope it won't cause an issue later on.
now the current error i'm getting is pretty much the same as for the OLED Library.
```
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::commonInit(const uint8_t*)':
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\Adafruit_ST7735.cpp:349:34: error: 'digitalPinToPort' was not declared in this scope; did you mean 'digitalPinToInterrupt'?
  349 |   csport    = portOutputRegister(digitalPinToPort(_cs));
      |                                  ^~~~~~~~~~~~~~~~
      |                                  digitalPinToInterrupt
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\Adafruit_ST7735.cpp:349:15: error: 'portOutputRegister' was not declared in this scope
  349 |   csport    = portOutputRegister(digitalPinToPort(_cs));
      |               ^~~~~~~~~~~~~~~~~~
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\Adafruit_ST7735.cpp:350:15: error: 'digitalPinToBitMask' was not declared in this scope
  350 |   cspinmask = digitalPinToBitMask(_cs);
      |               ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Raspberry Pi Pico.
```

i think i documented everything i did, but in case i missed something i can hopefully provide more detail.

earlephilhower, 2021-03-30:
Can you undo your library edits and try PR #59 and report back?

ProxyPlayerHD, 2021-03-30:
now i'm getting this error:
```
C:\Users\x\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\0.9.13\cores\rp2040/Arduino.h:28:10: fatal error: api/ArduinoAPI.h: No such file or directory
   28 | #include "api/ArduinoAPI.h"
      |          ^~~~~~~~~~~~~~~~~~
```

And I'm not sure if that is because i "installed" the test build wrong...
I removed the avr folder i added, and reverted the changes to the TFT Library.
then I downloaded the "avr" branch and replaced the files that got changed with the newly downloaded ones.

I'm sorry, i'm not that great at using github.

earlephilhower, 2021-03-30:
I assume you're using a `git clone`, yes?  In that case make sure you do the `git submodule update --init` in the main clone directory and in the pico-sdk directory (see the readme for examples).

ProxyPlayerHD, 2021-03-30:
no i just directly downloaded it from the site.
i'll try with git, but i'm running windows so the commands in the README.MD don't really work without modifying them.
speaking of which, why do they seem to download and install the board into the Arduino folder where your skteches are located `/home/earle/Arduino` instead of the folder that holds all the boards `home/earle/.arduino15`?
or am i missing something?

closed 2021-03-30 by earlephilhower

earlephilhower, 2021-03-30:
I went and build your examples from the original report and they built fine, so I just did a new release.  You can delete whatever copies you were working on and try 0.9.14 (I built, but did not test, since I don't have the exact LCD in question).

ProxyPlayerHD, 2021-03-30:
i'm sorry for the inconvenience.

i deleted the old board folder, and downloaded the newest release from the board manager and the OLED sketch compiled, but the TFT one didn't.
it's the same error as before.

```
In file included from C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\Adafruit_GFX.cpp:35:
C:\Users\x\Documents\Arduino\libraries\TFT\src\utility\glcdfont.c:3:10: fatal error: avr/io.h: No such file or directory
    3 | #include <avr/io.h>
      |          ^~~~~~~~~~
```
just to make sure i reinstalled the TFT Library, and even the whole Arduino IDE. but no change.
i must've missed something
i re

earlephilhower, 2021-03-30:
That TFT library is busted for 2020, sorry.  The `#ifdef` on top needs to include `!defined(ARDUINO_ARCH_RP2040)` and it also is using the **deprecated** port registers as **8-bit pointers** .  The RP2040 has 32b GPIO registers, and accessing it as an 8-bit address will really mess things up both on reads and especially on writes.

ProxyPlayerHD, 2021-03-30:
dammit, it's a built-in Library as well. i would've expected those to be up-to-date.
but since they want to add RP2040 support anyways they have to fix all built-in libraries as some point... hopefully.

some better news, the OLED Display is actually working, somewhat fast too.
though there must be a bottleneck in my program, even setting the I2C clock to 800kHz doesn't speed it up a lot. hnm

earlephilhower, 2021-03-30:
Still, good news!  You can try overclocking (up to 250 seems to work for everyonr so far) via the menus if it is Pico code that's the bottleneck.  Good luck!

ProxyPlayerHD, 2021-03-30:
no matter what i set the I2C clock, or the CPU's Clock to a frame basically always takes the same amount of time.

does the Wire Library even support different clock speeds for the RP2040? (datasheet says it can be anything between 0-1MHz.

earlephilhower, 2021-03-30:
`Wire.setClock(1000*1000)` will set I2C to run at 1MHz.  Default is 100KHz  (many devices do not run w/1MHz clocking).