earlephilhower/arduino-pico #60: overflow in call to pwm_config_set_clkdiv in analogWrite

smischny, 2021-03-30:
I believe you have a bug in setting the clock divider for the PWM.   I think you meant to divide the clock hz by the product of the analogFreq and analogScale, but instead it is multiplying it and it is overflowing:     

```
extern "C" void analogWrite(pin_size_t pin, int val) {
    if (!pwmInitted) {
        pwm_config c = pwm_get_default_config();
        pwm_config_set_clkdiv( &c, clock_get_hz(clk_sys) / 1.0 * (analogScale * analogFreq));
        pwm_config_set_wrap( &c, analogScale );
        for (int i=0; i<30; i++) {
            pwm_init(pwm_gpio_to_slice_num(i), &c, true);
        }
        pwmInitted = true;
    }
```

earlephilhower, 2021-03-30:
Awesome, @smischny .  You've found a bug and the solution.  Would you like to make a pull request so you'll get credit in the git log?

smischny, 2021-03-30:
I gave it a shot, as I've never done a pull request before.   

Thanks!

closed 2021-03-30 by smischny