earlephilhower/arduino-pico #64: Pi Pico not work on clock Higher than 275MHz

wivach, 2021-03-31:
The Pi Pico doesn't seem to run at a clock higher than 250 MHz. 
Systems used for testing the Arduino IDE 1.8.13 and Pi Pico version 0.9.14

Thank you

earlephilhower, 2021-03-31:
Anything that's overclocking, as noted in the menu, is not guaranteed and may not work.

The Pico is guaranteed to run at up to 133MHz, no more.

So, you got 2x the rated speed which seems like an amazing deal from the guys at the Raspberry Pi foundation. :)

closed 2021-03-31 by earlephilhower

guidol70, 2022-10-27:
@wivach 
You can check this YT-Video about overclock on the Pico (Mhz <-> Voltage):
https://www.youtube.com/watch?v=G2BuoFNLoDM

See also vreg.h = https://raspberrypi.github.io/pico-sdk-doxygen/vreg_8h_source.html

For 300Mhz we do need 1.15V core voltage - but changing the default voltage to 1.15 in vreg.h doesnt help with the arduino-core for the Pico

I tested 300Mhz with the blinking example and it doenst start :(

earlephilhower, 2022-10-27:
I don't recommend it and can't support it, but if you're trying to hack internal SDK headers you **can** do it on your own local `git` copy, but you'll then need to run the `tools/libpico/make-libpico.sh` script to rebuild the `libpico.a` files.

You probably will also need to use the `Generic` board definition with a `Generic SPI/4` divider on the flash to avoid it dying due to OC'ing the flash bus.

guidol70, 2023-10-15:
@wivach 
In the RP2040-HW Support (Arduino-Core) I edited the boards.txt 
and can get the normal Pico working stable with RunCPM up to 270Mhz:
```
rpipico.menu.freq.270=270 MHz (Overclock)
rpipico.menu.freq.270.build.f_cpu=270000000L
```

and the PicoW up to 260Mhz:
```
rpipicow.menu.freq.260=260 MHz (Overclock)
rpipicow.menu.freq.260.build.f_cpu=260000000L
```

Thats not much more, but could get achieved with the standard-voltage for the Pico(W).

[boards.txt](https://github.com/earlephilhower/arduino-pico/files/12909274/boards.txt)

wivach, 2023-10-15:
> @wivach In the RP2040-HW Support (Arduino-Core) I edited the boards.txt and can get the normal Pico working stable with RunCPM up to 270Mhz:
>
> ```
> rpipico.menu.freq.270=270 MHz (Overclock) [...]

> @wivach In the RP2040-HW Support (Arduino-Core) I edited the boards.txt and can get the normal Pico working stable with RunCPM up to 270Mhz:
>
> ```
> rpipico.menu.freq.270=270 MHz (Overclock) [...]

That's good news. Thank you very much.

XZCE, 2023-12-02:
For the Pico (not W) the 270MHz limit is due to the 2MB flash chip, you can use the "/4" instead of "/2" boot2 file version (boot2_w25q080_4_padded_checksum) to go a fair bit beyond 270MHz at the default voltage, set in boards.txt.

For Pico W, it's the pio state machine SPI used to communicate to the WiFi chip that's the issue, it has a divide by 2 default (which @260MHz means it starts to get flaky at around 130MHz).  You can divide by 3 instead of 2 which should allow overclocking up to around 390MHz (if you increase the voltage).  Mine max out at around 321MHz at the standard voltage, one of mine goes to 330MHz without increasing the voltage.

The divide by 2 #define code for the Pico W pio state machine is here:

https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c#L38

But it's pre-compiled into a library here by Earle, and he's already said he's not all that interested in making this work.