earlephilhower/arduino-pico #65: Request/hint: Timer library

omikron88, 2021-03-31:
Timer library that allows the management of  repeating timers.

Bodmer, 2021-04-07:
What features are you looking for?

The SDK examples are easy to prt to the Arduino sketch format.  There are [3 here](https://github.com/raspberrypi/pico-examples/tree/master/timer), do any of these satisfy your needs?

Bodmer, 2021-04-07:
Here is the SDK hello_timer example converted to run as a sketch. The others can be converted using the same approach.

```
/**
 * Copyright (c) 2020 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include <stdio.h>
#include "pico/stdlib.h"

/// \tag::timer_example[]
volatile bool timer_fired = false;

int64_t alarm_callback(alarm_id_t id, void *user_data) {
    Serial.printf("Timer %d fired!\n", (int) id);
    timer_fired = true;
    // Can return a value here in us to fire in the future
    return 0;
}

bool repeating_timer_callback(struct repeating_timer *t) {
    Serial.printf("Repeat at %d\n", micros());
    return true;
}

void setup() {
    Serial.begin(115200);
    while (! Serial);
    delay(1000);
    Serial.printf("Hello Timer!\n");

    // Call alarm_callback in 2 seconds
    add_alarm_in_ms(2000, alarm_callback, NULL, false);

    // Wait for alarm callback to set timer_fired
    while (!timer_fired) {
        tight_loop_contents();
    }

    // Create a repeating timer that calls repeating_timer_callback.
    // If the delay is > 0 then this is the delay between the previous callback ending and the next starting.
    // If the delay is negative (see below) then the next call to the callback will be exactly 500ms after the
    // start of the call to the last callback
    struct repeating_timer timer;
    add_repeating_timer_ms(500, repeating_timer_callback, NULL, &timer);
    sleep_ms(3000);
    bool cancelled = cancel_repeating_timer(&timer);
    Serial.printf("cancelled... %d\n", cancelled);
    sleep_ms(2000);

    // Negative delay so means we will call repeating_timer_callback, and call it again
    // 500ms later regardless of how long the callback took to execute
    add_repeating_timer_ms(-500, repeating_timer_callback, NULL, &timer);
    sleep_ms(3000);
    cancelled = cancel_repeating_timer(&timer);
    Serial.printf("cancelled... %d\n", cancelled);
    sleep_ms(2000);
    Serial.printf("Done\n");
}

void loop() {

}
/// \end::timer_example[]
```

earlephilhower, 2021-04-08:
I'm of 2 minds about this.

Using the SDK is fine, but not very Arduino-like.  It is, however, the easiest to implement since it's actually nothing on my end.

A simple `Alarm` class to wrap it up would be more in keeping w/the Arduino "way."  But then, if/when an Arduino.cc port comes out, it will be a source of incompatibility (but they might not support writing  SDK code, either)...

khoih-prog, 2021-05-15:
You can try this [**RPI_PICO_TimerInterrupt library**](https://github.com/khoih-prog/RPI_PICO_TimerInterrupt) to see if it's OK for your use case.

closed 2021-06-04 by earlephilhower