earlephilhower/arduino-pico #66: Great job! Do you have a plan to adjust it for Platformio?

vladkozlov69, 2021-03-31:
Hello Earle,

your codebase looks really great. Do you have a plan to adjust it for Platformio? i.e. adding board definition and platformio-specific packages/platforms?

earlephilhower, 2021-03-31:
@vladkozlov69 I don’t use Platform IO but I do know of it from my other Arduino core.

It would be great if you or another user sent a PR for this.  I know it’s not a big project, but doubt I’ll have time soon on my own to look at it.

jhmaloney, 2021-04-28:
Has anyone else taken this on? If not, I can give it a shot.

jhmaloney, 2021-05-04:
Unfortunately, after consulting with Valerii Koval from PlatformIO, I've concluded that this project is more than I care to take on, especially since I'd need to learn a lot about creating build scripts. I also found that PlatformIO does not want to support two Arduino Pico frameworks, so PlatformIO support for this package would be an ongoing community effort.

That said, PlatformIO does have a way to pull unsupported packages from GIT repositories, so adding PlatformIO support could still be useful. This community project has support for things that are not (yet?) in the official Arduino Pico framework, including the LittleFS and Servo libraries, which my project (MicroBlocks) depends on. My sincere thanks to Earle and others who have contributed to this package!

Here are the steps Valerii outlined, in case someone else wants to give this a go.

> This integration might need a substantial amount of effort. Here are the steps I think will be needed.
>
> First, the community package uses a custom toolchain with a very recent version of the newlib library, so it is not compatible with the toolchain used in the official Arduino core. That means that you will need to repack and upload to the PlatformIO registry all the necessary toolchains. (I assume there are also other auxiliary tools that should be uploaded (e.g. elf to uf2 converter, some python scripts etc.))
>
> Second -- and most important -- is the build script. The Arduino IDE build logic must be adapted the PlatformIO build system API. This isn't going to be an easy task. All that logic will go to a new python file in this folder https://github.com/platformio/platform-raspberrypi/tree/develop/builder/frameworks/arduino . You can take a look at similar scripts in the STM32 platform, for example https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino/mxchip.py
>
> Finally, you will need to implement some logic so users can switch between cores. Usually, we do this using the "build.core" option. Here is how it described in our docs for STM32:
https://docs.platformio.org/en/latest/platforms/ststm32.html#configuration
>
> And here is the code responsible for switching cores (look how we change packages, their versions, build scripts, etc)
https://github.com/platformio/platform-ststm32/blob/develop/platform.py#L41

earlephilhower, 2021-05-04:
@jhmaloney it may not be as bad as it sounds, but it's your call.

I based this core off of my work w/the ESP8266 Arduino core (https://github.com/esp8266/Arduino) and I have a feeling the actual build script there could be used here with minor mods.  The toolchain packaging AFAIK isn't much, either, as I'm using the same GCC and newlib there and it didn't seem to be a big deal to the PIO-esp8266 maintainer.

maxgerhardt, 2021-07-02:
I've opened the matching issue in https://github.com/platformio/platform-raspberrypi/issues/13. I can actually also help with the needed builder script and PR into this repo.. they are not too difficult to write, mostly all the same with some core-specific configuration options added mostly.

earlephilhower, 2021-07-02:
Cool, @maxgerhardt .  If there are any changes, let me know!  I've followed the ESP8266 tools and core packaging setup, which already supports it, so we should be more than half way there.

I'd do it myself, but I've never been able to get PIO working for me, except to build Marlin.  There, it "just worked."  But using a git repo instead of a d/l'd package, I could never get it to work...

maxgerhardt, 2021-07-14:
FYI, today I've forked the core and added a [builder script](https://github.com/maxgerhardt/arduino-pico/blob/master/tools/platformio-build.py), along with the needed platform modifications to use your [`toolchain-pico`](https://github.com/earlephilhower/pico-quick-toolchain/) and being able to do a core-selection in the `platformio.ini` in https://github.com/maxgerhardt/platform-raspberrypi. A test firmware repo is at https://github.com/maxgerhardt/pio-pico-core-earlephilhower-test.

The current state is that I can successfully build the firmware (with the linker script also generated by `simplesub.py`), but it won't yet run correctly -- doesn't re-initialize USB and doesn't even seem to reach `setup()` to do a LED blinky. I'm sure there's something minor missing, and I'll figure it out the next day.

There's also some ToDos left for being able to select the "partition table" / flash size options, and adding support for uploading files as e.g. supported in ESP8266 & ESP32.

earlephilhower, 2021-07-14:
Great news!  If you're feeling lucky, IIRC platformIO supports debugging and we distribute both GDB and OpenOCD in the tools tarball.  But that's definitely a "nice to have" not a "need to have!"  :)

Once your PR is in, I think it should be simple enough to add any platformIO specific packaging stages to the automated release process.  So when I push a tag to generate a release, the CI system will generate whatever else is needed for PIO, too.

maxgerhardt, 2021-07-14:
>IIRC platformIO supports debugging and we distribute both GDB and OpenOCD in the tools tarball. But that's definitely a "nice to have" not a "need to have!" :)

Debugging support is handled on the platform level (`platform-raspberrypi`), independent of the framework, (Would also be cumbersome to re-add the same debugging logic / commands for every framework.) There, ([here](https://github.com/platformio/platform-raspberrypi/blob/3c6cc84610e7004eed5980510862a38aa987a7c2/platform.py#L62-L98), [here](https://github.com/platformio/platform-raspberrypi/blob/3c6cc84610e7004eed5980510862a38aa987a7c2/builder/main.py#L169-L229)) it already has support for debugging via JLink (`JLink.exe`)  and OpenOCD (with `picotool`, `cmsis-dap`, `raspberrypi-swd` or any other debug adapter). So that's actually automatically already there and PlatformIO already has the packages for it (with GDB being sourced from `toolchain-pico` then). I'll of course verify this with my own Pico :)

maxgerhardt, 2021-07-14:
I am moving towards the final steps of the builder script refinements and platform code modifications. 

It is now working that a user can specify the sketch and filesystem size in the `platformio.ini`, via a e.g. `board_build.filesystem_size = 1M` parameter. From that, the linker script input variables are calculated just like in this core. 

Also I've added the code in the platform to create LittleFS images with the config that this core uses *and* the ability to flash those images via `picotool` to the device. It is all working pretty nicely :) 

[image]

The final steps are making it work via OpenOCD / JLink.exe upload methods and verifying that debugging works.

maxgerhardt, 2021-08-08:
I'm making progress regarding adding debugging support using Picoprobe, but it's not 100% working yet.

[image]

The stacktrace doesn't reach back to the actual user function (in the sketch I do `Serial.printf()` and the furthest in the stacktrace is `SerialUART::write()` and doesn't halt in e.g. `setup()` or `loop()`. Interestingly it sees both Cortex-M0+ cores as hardware threads and I can inspect them individually, with only Core 0 running a firmware for now.

PR #250 has been opened in the meantime for the builder script for this core.

JonnyHaystack, 2022-04-28:
To me, https://github.com/earlephilhower/arduino-pico/issues/501 seems like the biggest issue with using this on PlatformIO right now. Adafruit TinyUSB isn't working at all, meaning USB functionality is severely limited. I couldn't figure out a way of using Pico SDK tinyusb directly either.

Laserjones, 2022-05-08:
Sorry if this is not the right place for my issue (please let me know in that case).

I have tried to use the core in PlatformIO, but I get an error in the line `#include Arduino.h` that says:
```
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\Users\micro\Documents\PlatformIO\Projects\Amigo\src\main.cpp).C/C++(1696)
cannot open source file "stdint.h" (dependency of "Arduino.h")C/C++(1696)
```

Here's my **platformio.ini**:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.f_cpu = 133000000L
board_build.filesystem_size = 0.5m
platform_packages = 
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/i686-w64-mingw32.arm-none-eabi-ed6d983.220212.zip
```
Can anyone help with this?

maxgerhardt, 2022-05-08:
Which operating system are you using? The platformio.ini has a Windows-specific download link, if you have a different one you need to adapt it.

You can also try and force a redownload of the components by removing the `<home folder>/.platformio/.cache` and `<home folder>/.platformio/packages/toolchain-pico` folders and building the project again.

jhmaloney, 2022-05-08:
Here's what I use on Mac OS:

platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/maxgerhardt/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.2-a/x86_64-apple-darwin14.arm-none-eabi-9d55fd1.220202.tar.gz

Hope that helps!

Laserjones, 2022-05-08:
Sorry, I should have mentioned that I'm using Windows 10, 64-bit. I used the current link provided on _https://github.com/earlephilhower/pico-quick-toolchain/releases_.

> You can also try and force a redownload of the components by removing the <home folder>/.platformio/.cache and <home folder>/.platformio/packages/toolchain-pico folders and building the project again.

I emptied the two folders. When trying to rebuild, they were filled again as expected. But the error remains, and the **.pio\build\pico\FrameworkArduino** folder is still empty. I then even created a new project from scratch with the above **platformio.ini** and the default barebone **main.cpp** file, but the problem still persists.

So it looks like the toolchain is not copied into the project for some reason ...?

Laserjones, 2022-05-08:
> Here's what I use on Mac OS:
> 
> platform_packages = maxgerhardt/framework-arduinopico@https://github.com/maxgerhardt/arduino-pico.git

That's interesting, this is not the link stated by Earle in his documentation, which is https://github.com/earlephilhower/arduino-pico.git (is there a specific reason for this change?). Anyway, I now tried using your link instead, but it did not solve the problem either ...

Laserjones, 2022-05-08:
The IntelliSense C/C++ Configuration of PlatformIO says that there is no valid compiler path found. I can specify a path manually, but I have no idea where the compiler resides ...

jhmaloney, 2022-05-09:
> That's interesting, this is not the link stated by Earle in his documentation, which is https://github.com/earlephilhower/arduino-pico.git (is there a specific reason for this change?). Anyway, I now tried using your link instead, but it did not solve the problem either ...

I was an early user of using Earle's framework with platformio and I'm using the platform_packages that Earle and Max recommended at the time. There may be a better way to do things now, but this is still working for me.

When I was first getting the toolchain set up I got errors like the one's you report because I had the wrong toolchain link. It's been a while, but I believe I picked the toolchain for my operating system from:

https://github.com/earlephilhower/pico-quick-toolchain/releases

Good luck!

Laserjones, 2022-05-09:
Could it depend on the way I created the original project before I adapted the **platformio.ini**? When I selected the board, there were several entries called "Raspberry Pico" or similar. I have no idea what the differences are, so I just picked the first one ...

maxgerhardt, 2022-05-09:
`board = pico` is perfectly valid. This is in fact the very first board in my [example](https://github.com/maxgerhardt/pio-pico-earlephilhower-test-all-variants/blob/main/platformio.ini).

Laserjones, 2022-05-09:
> `board = pico` is perfectly valid. This is in fact the very first board in my [example](https://github.com/maxgerhardt/pio-pico-earlephilhower-test-all-variants/blob/main/platformio.ini).

I know, but my thought was that besides the **platformio.ini** file, there might be other parts of the project set up differently depending which of the "Rapberry Pico" board entries I select in advance? Or does the **platformio.ini** control everything?
[image]

maxgerhardt, 2022-05-09:
The `platformio.ini` controls all base build configuration options, nothing else you selected in the UI matters. In fact one can create an "Arduino Uno" project then change the `platformio.ini` to the one you posted above and it will compile for a pico. No traces of an Arduino Uno then.

Laserjones, 2022-05-09:
> The `platformio.ini` controls all base build configuration options, nothing else you selected in the UI matters. In fact one can create an "Arduino Uno" project then change the `platformio.ini` to the one you posted above and it will compile for a pico. No traces of an Arduino Uno then.

Good to know. But then I have absolutely no idea why the compiler is not found ... Shouldn't it be installed along with the other stuff? Where would it usually be located?

maxgerhardt, 2022-05-09:
Is the folder `<home folder>/.platformio/packages/toolchain-pico` still **empty** after that? Even after removing `toolchain-pico` and that `.cache` folder referenced above? 

Do you have antivirus active? 

The only two possibilities I see is a corrupted download or an AV blocking execution.

Laserjones, 2022-05-09:
No, as I wrote above, the `toolchain-pico` and `.cache` folders were re-filled automatically after I emptied them. What is still empty is the `.pio\build\pico\FrameworkArduino` folder within the project.

I also found now that the compiler EXEs are indeed there in `toolchain-pico\bin`. I tried to set this path as the compiler path in the IntelliSense settings (which I can open via the light bulb icon next to the error message in the `#include Arduino.h` line, which has reappeared for some reason), but for some reason it is not saved. Is there a config file I can edit directly? I found a website that speaks of a  `.vscode\c_cpp_properties.json` file including the compile path, but it does not exist on my system.

I'm using the standard Windows Defender as AV. I don't think it has to do with that. I could compile a project using the MBED core without problems.

maxgerhardt, 2022-05-09:
>tried to set this path as the compiler path in the IntelliSense settings (which I can open via the light bulb icon next to the error message in the #include Arduino.h line, which has reappeared for some reason), but for some reason it is not saved.

PlatformIO controls + will overwrite the `.vscode/` configuration files, it has to be fixed in PlatformIO, VSCode follows automatically.

>I also found now that the compiler EXEs are indeed there in toolchain-pico\bin. 

What is your exact OS + processor type? Windws 64-bit or 32-bit? 

Can you try adding an exception for the `toolchain-pico/bin` folder in Windows defender per https://support.microsoft.com/en-us/windows/add-an-exclusion-to-windows-security-811816c0-4dfd-af4a-47e4-c301afe13b26?

Laserjones, 2022-05-09:
> Can you try adding an exception for the `toolchain-pico/bin` folder in Windows defender 

This did not help.

> What happens when you open a terminal (Windows + R -> cmd.exe), go into the compiler directory (`cd <path to toolchain-pico/bin` folder) and execute `arm-none-eabi-gcc --version`?

I get the same error message as in the compiler terminal (which explains why it is German, as it is generated by Windows directly). When I run the terminal as Administrator, the command works (version is displayed: `arm-none-eabi-gcc (GCC) 10.3.0`). However, running VSCode as Administrator did **not** help.

maxgerhardt, 2022-05-09:
There's something really wrong when you can execute a binary which should have been created as a regular user only as administrator, and not as a regular user. :/. 

At this point I can only propose a clean reinstall of PlatformIO *as a user*. That is, remove the VSCode PlatformIO extension, remove the entire `<home folder>/.platformio` folder, re-open VSCode, reinstall the PlatformIO extension, re-import your old Pico project and try compiling it again.

Laserjones, 2022-05-09:
It's getting stranger and stranger. After reinstalling PlatformIO, I can now call the compiler via the terminal as a normal user, but trying to compile the project, the error messages still appear!

Should I reinstall VSCode from scratch?

maxgerhardt, 2022-05-09:
It may be worth a try, as I'm out of ideas 😄.

Laserjones, 2022-05-11:
No other extensions. For now I have given up and switched to the Arduino IDE 2.0.

vaelen, 2022-05-24:
I'm having a weird, but very minor, issue on Linux with PlatformIO. It reinstalls the toolchain every time I build.
```
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/arm-linux-gnueabihf.arm-none-eabi-ed6d983.220212.tar.gz
Unpacking  [########----------------------------]   23%  00:00:38
```

Happy to move this to a separate issue too.

PS. Fantastic work on this core and on the PlatformIO integration.

vaelen, 2022-05-24:
Yes, it still does it on the newest build.
```
andrew@raspberrypi:~ $ pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 6.0.2a2
Release notes: https://docs.platformio.org/en/latest/history.html
```

The file does exist. The contents are listed below.
```
andrew@raspberrypi:~ $ cat ~/.platformio/packages/toolchain-pico/.piopm 
{"type": "tool", "name": "toolchain-pico", "version": "5.100300.220212", "spec": {"owner": "maxgerhardt", "id": null, "name": "toolchain-pico", "requirements": null, "uri": "https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/arm-linux-gnueabihf.arm-none-eabi-ed6d983.220212.tar.gz"}}
```

Here is my platform
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower

platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/arm-linux-gnueabihf.arm-none-eabi-ed6d983.220212.tar.gz
```

maxgerhardt, 2022-05-24:
Could you try using `https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/arm-linux-gnueabihf.arm-none-eabi-7855b0c.210706.tar.gz` as the download link for toolchain-pico? I have tested that against an earlier version. Compilation might fail, but if you compile it multiple times, does it unpack the toolchain anew every time?

vaelen, 2022-05-24:
Same behavior with that version.  I'll see if I can setup another project with a similar toolchain requirement, where it links to a specific URL instead of just a version. It might be a bug in `pio` on Linux.

maxgerhardt, 2022-06-09:
We did a major integration rehault at https://github.com/platformio/platform-raspberrypi/pull/36, please update the platform as described in the third comment of the PR and use the same platformio.ini as the PR (no more `platform_packages`). Does it still occurr?

vaelen, 2022-06-10:
That fixed it! Thanks. :)

maxgerhardt, 2022-07-15:
Please just use
```ini
[env:nanorp2040connect]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = nanorp2040connect
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
platform_packages = 
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
```
And use [the CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) to `pio pkg update -g -p https://github.com/maxgerhardt/platform-raspberrypi.git`.

Your PlatformIO installation seems to want to install an outdated compiler version (`5.100300.220629`)  when a new one was released just [4 hours ago](https://registry.platformio.org/tools/earlephilhower/toolchain-rp2040-earlephilhower/versions). If these caching issues persist, please delete `C:\Users\<user>\.platformio\platforms\raspberrypi*` and rebuild the project.

earlephilhower, 2022-07-15:
Yeah, @maxgerhardt , I was just about to drop a note in the PR about the updated toolchain which was just published.  Was debugging the PicoW LWIP stack when I found I missed a mutex in the `FILE` structure, so I needed a new toolchain build.

NicolasDumoulin, 2022-07-15:
Thanks @maxgerhardt for your very quick answer!

I've updated the CLI for the mentionned update and removed the cache dirs ~/.platformio\platforms\raspberrypi* and .platformio/.cache

But I've still the error when I try to build the project:
```
Processing nanorp2040connect (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: nanorp2040connect; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/maxgerhardt/platform-raspberrypi.git
git version 2.25.1
Clonage dans '/home/dumoulin/.platformio/.cache/tmp/pkg-installing-_cgm83ha'...
remote: Enumerating objects: 70, done.
remote: Counting objects: 100% (70/70), done.
remote: Compressing objects: 100% (57/57), done.
remote: Total 70 (delta 32), reused 25 (delta 2), pack-reused 0
Dépaquetage des objets: 100% (70/70), 121.78 Kio | 733.00 Kio/s, fait.
Sous-module 'builder/frameworks/arduino/mbed-core' (https://github.com/platformio/builder-framework-arduino-core-mbed.git) enregistré pour le chemin 'builder/frameworks/arduino/mbed-core'
Clonage dans '/home/dumoulin/.platformio/.cache/tmp/pkg-installing-_cgm83ha/builder/frameworks/arduino/mbed-core'...
remote: Enumerating objects: 31, done.        
remote: Counting objects: 100% (31/31), done.        
remote: Compressing objects: 100% (23/23), done.        
remote: Total 31 (delta 8), reused 24 (delta 7), pack-reused 0        
Chemin de sous-module 'builder/frameworks/arduino/mbed-core' : 'ed9a3458e336660fdf2b1d4a4c828a2ddfd214ad' extrait
Platform Manager: raspberrypi@1.7.0+sha.28e5339 has been installed!
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220629
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220629' requirements for your system 'linux_x86_64'
```

maxgerhardt, 2022-07-15:
My bad, I actually hardcoded the toolchain reference with its datecode and so naturally I had to update it. Please issue the `pio platform update ...` command again and rebuild.

NicolasDumoulin, 2022-07-15:
Cool. It works ✌️

JonnyHaystack, 2022-07-18:
After updating the platform just now, I'm getting the following error
```
 *  Executing task: C:\Users\Haystack\.platformio\penv\Scripts\platformio.exe run --environment pico 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi; framework: arduino; board: pico)
----------------------------------------------------------------------Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-rp2040-earlephilhower@5.100300.220714 has been installed!
Tool Manager: Installing earlephilhower/framework-arduinopico @ ~1.20202.0
Error: Could not find the package with 'earlephilhower/framework-arduinopico @ ~1.20202.0' requirements for your system 'windows_amd64'
```

maxgerhardt, 2022-07-19:
Alright I just went ahead and updated a bunch of stuff, which as a sideffect gives us Raspberry Pi Pico W support, see [PR](https://github.com/platformio/platform-raspberrypi/pull/36).

WaGi-Coding, 2022-07-21:
I finally managed to upload, but unfortunately it cannot detect any networks, while it works fine with micropython.

Thats my main.cpp
```cpp
#include <Arduino.h>
#include <WiFi.h>

void setup() { Serial.begin(115200); }

void loop() {
  Serial.println(WiFi.scanNetworks());
  delay(5000);
}
```

Thats my platformio.ini
```ini
[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower

platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-c/x86_64-w64-mingw32.arm-none-eabi-0196c06.220714.zip
```

Thats the output:

```
0

--- Sleeping ---

0

--- Sleeping ---

0

--- Sleeping ---
```

With micropython i find a bunch of networks.
VSC does not show any errors or warnings, i am lost ^^

earlephilhower, 2022-07-21:
I just testes and the ScanNetworks example works fine, so I think you have a build problem.  I'd dump the full compile commands and make sure you have the proper defines set.

````
Beginning scan at 188393
Found 9 networks

                            SSID   ENC     BSSID         CH RSSI
                                  WPA2 08:36:C9:87:C8:FF  9  -86
                                  WPA2 10:0C:6B:50:2D:EB  6  -92
                        JK-Guest  WPA2 12:36:C9:87:C8:FF  9  -86
                           Scout  WPA2 30:FD:38:FD:69:E2  6  -90
                        NOBABIES  WPA2 38:2C:4A:90:FA:80 10  -32
                       Grace4all  WPA2 4C:B1:CD:0D:2C:F8  6  -89
                        NOBABIES  WPA2 50:D4:F7:FE:A3:4D  1  -73
             ngHub_31946CN402774  WPA2 9C:3D:CF:F3:89:AF 11  -88
                        NOBABIES  AUTO AC:84:C6:6C:EF:F3  1  -60

--- Sleeping ---
````

A normal IDE compile line looks like:
````
/home/earle/.arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/arm-none-eabi-g++ -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0xf00a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=0 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/home/earle/Arduino/hardware/pico/rp2040/ @/home/earle/Arduino/hardware/pico/rp2040/lib/platform_inc.txt -I/home/earle/Arduino/hardware/pico/rp2040/include -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0xf00a -DUSBD_MAX_POWER_MA=250 -DF_CPU=133000000L -DARDUINO=10816 -DARDUINO_RASPBERRY_PI_PICO_W "-DBOARD_NAME=\"RASPBERRY_PI_PICO_W\"" -DARDUINO_ARCH_RP2040 -Os -I/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040 -I/home/earle/Arduino/hardware/pico/rp2040/variants/rpipicow -I/home/earle/Arduino/hardware/pico/rp2040/libraries/WiFi/src /home/earle/Arduino/hardware/pico/rp2040/libraries/WiFi/src/BearSSLHelpers.cpp -o /dev/null
````

WaGi-Coding, 2022-07-21:
Hmm, I have no idea how i'd check that to be honest. I just got into platformio and i think it's just too much for me it seems & i should prob. stick to micropython or wait for official support :S

Sorry for my confusing

maxgerhardt, 2022-07-21:
I will double check the compile commands on PlatformIO -- since I don't have my Pico W yet, I couldn't test on real hardware, only got it to *compile*. Though the adaptions of the builder script made before me looked perfectly fine.

WaGi-Coding, 2022-07-21:
> I will double check the compile commands on PlatformIO -- since I don't have my Pico W yet, I couldn't test on real hardware, only got it to _compile_. Though the adaptions of the builder script made before me looked perfectly fine.

If there is any way i can remotely help you with it easily, feel free to tell me/us what to do ^^ Also, if you really need one(they seem to be hard to get atm), i can send you one of mine from Austria to Germany. Free of charge as thanks for your work

maxgerhardt, 2022-07-21:
In the meantime @WaGi-Coding, can you install [the Arduino-Pico core in the Arduino IDE](https://github.com/earlephilhower/arduino-pico#installing-via-arduino-boards-manager) and see if you also get 0 networks?

maxgerhardt, 2022-07-21:
Okay, thanks for confirming, if the Arduino IDE and PlatformIO results differ with the exact same code, then the fault is likely in the PIO builder script -- I'll have a look.

>Also, if you really need one(they seem to be hard to get atm), i can send you one of mine 

I preordered two Pico W from Elektor the day the sent the email that preorders are available. And wow, looking at [their website](https://www.elektor.com/raspberry-pi-pico-rp2040-w) now it used to say available on July 20th, now it says August 1st.. not a good sign :/. I hope they're really shipped to me by then.

maxgerhardt, 2022-07-21:
@WaGi-Coding I have a bad dejavu with variant-specific `initVariant()` not being linked correctly due to using a `.a` archive instead of an object file. In the code for PlatformIO, can you use the code
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <pico/cyw43_arch.h>

void setup() { cyw43_arch_init(); Serial.begin(115200); }

void loop() {
  Serial.println(WiFi.scanNetworks());
  delay(5000);
}
```
and retry?

WaGi-Coding, 2022-07-21:
@maxgerhardt oh yeah that seems to work, many many thanks <3 
[image]

maxgerhardt, 2022-07-21:
Alright then I know **exactly** what's wrong, many thanks. I'll PR this in shortly.

maxgerhardt, 2022-07-21:
This will be fixed by https://github.com/earlephilhower/arduino-pico/pull/696.

maxgerhardt, 2022-07-21:
@WaGi-Coding since the PR was merged and I updated the commit reference (https://github.com/maxgerhardt/platform-raspberrypi/commit/bd2f3e756ba7340150f2bebc2d46110988243018), please do a `pio pkg update -g -p https://github.com/maxgerhardt/platform-raspberrypi.git` in [the CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and try it without calling into `cyw43_arch_init();` explicitly -- Arduino IDE and PlatformIO behavior should be exactly equal now.

WaGi-Coding, 2022-07-21:
@maxgerhardt it works like a charm now. Again, many thanks

SirSydom, 2022-07-25:
We from [OpenKNX](https://github.com/OpenKNX) are impatiently waiting for this getting into offical PIO because we want it to be our default Toolchain.
What is the status of the pio integration, since some week it looked to me that it may be only a matter of days  ?
Thanks.

maxgerhardt, 2022-07-25:
The holdup with the merge into official PIO is at https://github.com/platformio/platform-raspberrypi/pull/36, not in this issue; I can't say what is holding it up directly, but I'm trying to resolve it soon. Had a lot to do the past weeks, and this looks like it's going to be a bit more involved..

[55 more replies]

closed 2023-04-08 by earlephilhower