earlephilhower/arduino-pico #70: How to use multicore_launch_core1();

beegee-tokyo, 2021-04-02:
Hello @earlephilhower 

Great work you are doing here, really appreciated.
I am just starting with the Pico and I guess I make a stupid mistake.
In the [Pico Examples](https://github.com/raspberrypi/pico-examples) is some code how to use the multicore functions.
But when I try to `#include "#include "pico/multicore.h"` compilation fails with file not found. Am I missing something?

If it is not yet implemented here, that is ok, I can wait (or try to implement and do a PR). Just wanted to know before I put time into it.

earlephilhower, 2021-04-02:
Please use 0.9.16, just released. It was just a matter of adding the proper include path to the platform file.  Good luck!

closed 2021-04-02 by earlephilhower

beegee-tokyo, 2021-04-03:
@earlephilhower 

Thank you. Will update today and give you feedback.