earlephilhower/arduino-pico #76: LittleFS format() hangs with flash size > 2MB

blurfl, 2021-04-03:
Trying out LittleFS/SpeedTest on a Feather, the program never returns from format(). Changing the board type to "Generic RP2040  Flash Size 2MB with Boot Stage 2: W25X10CL QSPI /4" works as expected, but upping to Flash Size 4MB fails.

earlephilhower, 2021-04-03:
Unfortunately I don't have anything but the Rpi Pico 2MB to test on.  FWIW the LittleFS library runs on the ESP8266 just fine w/a 3MB FS on a 4MB flash chip, so I don't think it's something there.

The fussy flash on the Feather might have some special needs for programming, not just booting...

Do you have a Picoprobe to run GDB on your board with?  Even just connecting and getting a stack dump would help pinpoint what's going on./

earlephilhower, 2021-04-03:
Can you test the EEPROM examples with the normal 8MB settings?  They also access the flash using the same method as the FS.  If they pass but the FS fails it might be something other than flash...

blurfl, 2021-04-03:
The eeprom_write example fails (hangs) for any of the 4MB or 8MB flash settings. Sorry, no Picoprobe here  yet.

earlephilhower, 2021-04-03:
The SDK was refusing to write past what it thought end of flash was.  Latest release should sort it.

closed 2021-04-03 by earlephilhower

blurfl, 2021-04-03:
The latest release does take care of it.👍

commit c1ab2b73 (2021-04-03): Update FLASH_SIZE_BYTES to avoid assert on > 2MB

The SDK would assert if a flash write/erase happened past the end of the
flash space it was built with.  So, 8MB chips would crash when accessing
offset 3MB, for example.

Avoid this by specifying the maximum flash available so the SDK won't
trip.

Fixes #76