earlephilhower/arduino-pico #77: Does not work with Arduino IDE 2.0

Lucvolders, 2021-04-03:
There is an issue woth Arduino IDE 2.0.
Versions would not go higher as 0.9.9

So I installed newest versions with the old Arduino IDE
installed version 0.9.18 works ok there but now the Pico
does not show up in the boards manager at all anymore in 2.0.

Do not know if it is a problem on your side or on the arduino side.....

earlephilhower, 2021-04-03:
That looks to be a bug in their 2.0 beta.  https://semver.org shows that you're allowed 1 or more digits between the dots.  (If not, you would never be able to release version 10 of anything!).  I'd recommend opening up a bug in their repo.

earlephilhower, 2021-04-03:
Oh, and this shouldn't be an issue for long.  Now that I have all the features I wanted in, it's just a matter of a few docs and bug fixes before 1.0.0. :)

closed 2021-04-03 by earlephilhower