earlephilhower/arduino-pico #95: U2conv.py error on Windows10

al1fch, 2021-04-08:
Hi
No problem with Ubuntu, works great !! Thanks !!
 
 I got u2conv.py error when i try  to flash Pico Py board  
Windows 10 French
Arduino IDE 1.8.13
Raspberry Pi Pico core vesrion 1.0.1
Same error with Fade or any other example code
Error message :

```
Resetting COM8
Traceback (most recent call last):
Une erreur est survenue lors du transfert du croquis
  File "C:\Users\Lenovo\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.0.1/tools/uf2conv.py", line 368, in <module>
Converting to uf2, output size: 439296, start address: 0x2000
    main()
  File "C:\Users\Lenovo\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.0.1/tools/uf2conv.py", line 355, in main
    drives = get_drives()
  File "C:\Users\Lenovo\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.0.1/tools/uf2conv.py", line 233, in get_drives
    for line in to_str(r).split('\n'):
  File "C:\Users\Lenovo\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.0.1/tools/uf2conv.py", line 225, in to_str
    return b.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 225: invalid continuation `byte`
```

Best Regards

earlephilhower, 2021-04-08:
Somehow `wmic` is reporting something very weird on your system.  each drive entry should be 2-3 bytes (i.e. `C:`), not >200, and there shouldn't be anything outside plain ASCII.

Can you open a cmd window and run `wmic path win32_logicaldisk get deviceid, volumename, filesystem, drivetype` and post the output to let us see what the script is seeing?

Bodmer, 2021-04-08:
I suspect a volume name has an accented character. Although these are in the range 128-255, some in that range get utf8 encoded.

al1fch, 2021-04-09:
Thanks !  Solved 
No bug ,the problem was on my side : **µ** character in a volume 
Here is the result of mentionned windows cmd :

```
DeviceID  DriveType  FileSystem  VolumeName
C:        3          NTFS
D:        3          NTFS
E:        3          FAT32
F:        2          FAT32       µSD
```

Bodmer, 2021-04-09:
Makes sense, mu is 0xE6 in the French extended character set. So the volume name is reported in pure ASCII but is being interpreted in Python as utf8 encoded, and then it is an illegal sequence. So it is a bug, since the extended character set is allowed in Windows volume names.

earlephilhower, 2021-04-09:
Well, not pure ASCII as that would only have from 0..127, but in the Windows codepage for French which is definitely not UTF-8!

It's a pain because the upload tool is itself from Microsoft(!!!) and they mixed up codepages and utf-8.  If SO is correct, WMIC uses the codepage for piped outputs but changes to Unicode if redirected to a file.

earlephilhower, 2021-04-09:
Oh, and thanks @al1fch for following up.  Now that I know how to repro on Windows, I can try and find a solution.  Powershell, maybe, as I think that's always unicode...

PR #148 (2021-05-19): Ignore non-UTF8 chars in UF2 upload
Fixes #95
commit 0d593252 (2021-05-19): Ignore non-UTF8 chars in UF2 upload (#148)

Fixes #95
closed 2021-05-19 by earlephilhower