earlephilhower/arduino-pico #96: Windows 7 sometimes does not recognize the device as serial port

PsychoXIVI, 2021-04-09:
Windows 7 sometimes does not recognize the device as serial port, even after fresh reinstalling using BOOTSEL button. You still can upload sketch using BOOTSEL mode manually, but it requires disconnecting device each time. Simply, the issue disable easy upload and auto-restart feature. 

There is "ArduinoPico" unrecognized device in the device manager list with ID: `USB\VID_2E8A&PID_000A&REV_0100` or similar. I am using Arduino IDE 1.8.13. Using different USB port didn't help. Restarting the IDE didn't help. On Windows 10 the same board works without any problems.

I found solution to make it work also on Windows 7:
[image]
(from https://www.onetransistor.eu/2021/02/set-up-raspberry-pi-pico-for-microphyton.html , btw: I used "Install driver")

Tested on Windows 7 that had previously this issue. Maybe you can add it to README or somewhere?

prenticedavid, 2021-04-10:
Thanks for the tip.   I have a Win7-32 laptop.   This gets a Serial Terminal and Program Upload.    Opening the Arduino Serial Terminal does **not** reset the Pico.
If you upload when the Serial Terminal is open you tend to get an Error when it resets baud rate.
If you upload with a Closed Serial,   you need the program to wait e.g. with `while (!Serial) ;`

But my big problem is Win10-64 Desktop PC.
I can upload if I put the Pico into Boot mode.
But the Serial will only run if I toggle the power.

After toggling power I see PicoArduino (COM23) in the Device Manager.
Properties say: usbser.sys
Driver Version: 1.0.0.0
Digital Signer: USB\VID_2E8A&PID_000A (libwdi autogenerated)

Has Win10 installed the correct Driver ?

David.

closed 2021-04-14 by earlephilhower

Andy2No, 2021-05-09:
I've just gone through this process, and that wasn't exactly what I saw, so here are some screenshots which might make it clearer to someone else.  The serial port was shown as PicoArduino.

Zadig (1) - Immediately after plugging in the Pi Pico, Zagig sees PicoArduino which shows as no driver, WinUSB

[image]

Zadig (2) - PicoArduino (2E8A) needs to be set to USB Serial (CDC), using the arrow keys next to the driver type, then pressing Install Driver.

[image]

Zadig (3) - PicoArduino (2E8A) after pressing Install Driver it looks like this:

[image]

commit d490499c (2021-04-14): Add note about Win7 driver installation (#102)

Fixes #96
PR #102 (2021-04-14): Add note about Win7 driver installation
Fixes #96