earlephilhower/arduino-pico #99: Cheap I2S input test Mics.

jake653, 2021-04-12:
FYI

The Microcenter Stores cleared out the original Google Voice Kits for $2.99, if you know anyone with one, the I2S mic. (SPH0645LM4H-B) pair connector is:

Blk   Gnd
Grn  I2S DIN
Wht I2S BCLK 
Blu   I2S LRCLK
Red  3.3V

earlephilhower, 2021-04-14:
Thanks for the note.  I actually have a couple of them from my ESP8266 work.

Unfortunately the PIO program needs to be rewritten to support I2S input.  It's in the queue but not very high priority.  If someone has something to contribute to add it sooner, that would be great.

giampiero7, 2022-02-22:
Hi,
I've been working on this, putting together what I've found around.

This module by @kilograham in the official [pico-extras repo](https://github.com/raspberrypi/pico-extras) also supports output only:
https://github.com/raspberrypi/pico-extras/tree/master/src/rp2_common/pico_audio_i2s

The most complete support for I2S in/out that I've found is this MicroPython library by @miketeachman:
https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2s.c
with great examples and documentation here:
https://github.com/miketeachman/micropython-i2s-examples

From there I've extracted the parts i needed (I2S input, blocking) and wrapped it in a Arduino-Pico-like library:
https://github.com/sfera-labs/arduino-pico-i2s-audio
It works, more tests are needed.

Now, it would be great to have support directly in the standard pico SDK or at least in the pico-extras.
I think the way to go would be to port @miketeachman's code in a C pico library and then use that to wrap it the MicroPython and Arduino-Pico libraries.

@miketeachman, @kilograham, @earlephilhower what do you think? Can we work on it?

All the best,
Giampiero

earlephilhower, 2022-02-24:
Nice work, @giampiero7.  This core now just uses the pico-extras I2S code with a thin `class I2S` wrapper, FYI.  Looking at the micropython code it doesn't actually look that difficult (and I can see how to optimize it a bit plus add support for 24-bit samples).

You might want to consider doing a PR against pico-extras w/an addition. I'm not sure about the licenses (MIT vs. BSD-3), though.

miketeachman, 2022-02-24:
Glad the you can use the MicroPython implementation to get better I2S support in the Arduino rp2 implementation.  I don't see any licensing problems..  You'll just need to keep the MIT licensing comment as you have used "substantial portions of the Software" as outlined in the MIT license.  I'll keep an eye out for any significant improvements or bugs you find.  I will be adding MCK support to the MicroPython rp2 port at some point as per this [PR-I2S on mimxrt port ](https://github.com/micropython/micropython/pull/8322).

giampiero7, 2022-02-24:
Hi @miketeachman,
Glad to have your feedback and thanks for your work.
Given that it is essentially all your code, would you be willing to start a PR against pico-extras with your module stripped off the MicroPython overhead (on the line of what I've done [here](https://github.com/sfera-labs/arduino-pico-i2s-audio/blob/master/src/machine_i2s.c))? So that everything else (including the MicroPython module) can build around it, instead of having different code for each language...

BTW, I'm running some tests and noticing some strange data coming from my mic... I'll be bothering you with an issue on your repo tomorrow probably...

miketeachman, 2022-02-24:
> Given that it is essentially all your code, would you be willing to start a PR against pico-extras with your module stripped off the MicroPython overhead (on the line of what I've done [...]

@giampiero7 I really like your idea of a shared I2S codebase, but I'm not sure that pico-extras is the best place.  The code in pico-extras is somewhat experimental as highlighted by the warning "Note that any API here is a work in progress and subject to change".  Perhaps consider adding I2S support to the Pico SDK, e.g.  pico-sdk/tree/master/src/rp2_common/hardware_i2s.

A challenge for my involvement is that I don't have any time to work on this initiative.  Any volunteer development time goes towards MicroPython.  Sorry.

> BTW, I'm running some tests and noticing some strange data coming from my mic... I'll be bothering you with an issue on your repo tomorrow probably...

Great.  I like to know about any problems.  It would help me to have a minimal code example that can reliably reproduce the issue, ideally written in MicroPython as I rarely work in the Arduino environment.  [Here is a blocking example in MicroPython](https://github.com/miketeachman/micropython-i2s-examples/blob/master/examples/record_mic_to_sdcard_blocking.py)

Thanks!

giampiero7, 2022-02-25:
> @giampiero7 I really like your idea of a shared I2S codebase, but I'm not sure that pico-extras is the best place. The code in pico-extras is somewhat experimental as highlighted by the warning [...]

I just thought pico-extras would be a good test area, to be later promoted to the SDK once the API is finalized.

> A challenge for my involvement is that I don't have any time to work on this initiative. Any volunteer development time goes towards MicroPython. Sorry.

Well, being it a common codebase, it would still be a contribution to MicroPython after all ;P 

Anyway, I'll go on with the tests and then I'll find a way to persuade you :)

commit f1e28ab5 (2022-05-02): Rewrite I2S from scratch, add input support

Rewrite the I2S code from scratch to eliminate the dependence on the
pico-extras implementation and to support I2S input as well.

8-bit, 16-bit, 24-bit, and 32-bit words are supported.

Multiple I2S ports are allowed.

I2S input and I2S output are supported.

Fixes #535
Fixes #99
PR #569 (2022-05-05): Rewrite I2S from scratch, add input support
Rewrite the I2S code from scratch to eliminate the dependence on the
pico-extras implementation and to support I2S input as well.

8-bit, 16-bit, 24-bit, and 32-bit words are supported.

Multiple I2S ports are allowed.

I2S input and I2S output are supported.

Fixes #535
Fixes #99
commit 07500e89 (2022-05-05): Rewrite I2S from scratch, add I2S input support (#569)

Rewrite the I2S code from scratch to eliminate the dependence on the
pico-extras implementation and to support I2S input as well.

8-bit, 16-bit, 24-bit, and 32-bit words are supported.

Multiple I2S ports are allowed (theoretically up to 6 because
2 DMA channels are required per port).

I2S input and I2S output are supported.

Add input example

Fixes #535
Fixes #99
Fixes #562
closed 2022-05-05 by earlephilhower