earlephilhower/arduino-pico #104: NOT_AN_INTERRUPT not #defined

obdevel, 2021-04-15:
I have a 3rd party library that checks whether digitalPinToInterrupt() returns NOT_AN_INTERRUPT, which isn't defined in this core. 

Clearly, all Pico pins are interrupt-capable, but it would still be useful to have this #defined. The AVR core #defines it in Arduino.h as (-1).

I have hacked the library as a workaround so there is no rush for a solution.

If this is an unsurmountable problem, I can ask the library author to handle the Pico as a special case.

Thanks.

earlephilhower, 2021-04-15:
Latest core now has this define set.  You can use git or just add the 1-line(like I think you already did)., or wait for 1.04 in a little bit.

closed 2021-04-15 by earlephilhower

obdevel, 2021-04-15:
Thank you.

There is no rush so I can wait for 1.0.4.

It'll be interesting to see how support for this board and chip develops. Personally, I'd like to see greater support for multiprocessing and maybe an RTOS port. Hopefully the bare chips will be available soon too.

Everything I have run so far on your core 'just works'. I plan to investigate playing out sound files using the I2S library.

Best wishes from Cambridge, UK ... home of RPF and Arm :)

Duncan
e: ***@***.***

PR #105 (2021-04-15): Add NOT_AN_INTERRUPT define
Fix #104
commit b5aeb84c (2021-04-15): Add NOT_AN_INTERRUPT define (#105)

Fix #104