earlephilhower/arduino-pico #112: Installing via GIT

reppr, 2021-04-22:
To install via GIT (for latest and greatest versions):

mkdir -p ~/Arduino/hardware/pico
git clone https://github.com/earlephilhower/arduino-pico.git ~/Arduino/hardware/pico/rp2040
cd ~/Arduino/hardware/pico/rp2040
git submodule update --init
cd pico-sdk
git submodule update --init
cd pico-extras                  <<<<<<    this line should be  'cd ../pico-extras'   >>>>>>>
git submodule update --init
cd ../tools
python3 ./get.py

reppr, 2021-04-22:
I have two related questions:

1) while the installation from git works well (with the above path correction) it is not clear to me how to make the boards visible in the Arduino IDE. After restarting Arduino I do not see any rp2040 boards to select them. Installing again with the boards manager does not let me select the git version.
So I am confused about how to change from an installation by the boards manager to an installation from git

2) If I will do an update from git after a while will a simple git pull also work on submodules?

Thanks you for advice

earlephilhower, 2021-04-22:
Thanks for the note, it's been fixed.

1) You need to git clone from in: <Arduino sketch directory>/hardware/pico/.  On Linux, it's ~/Arduino/hardware/pico/.  Windows/Mac will be elsewhere but similar.  So I have ~/Arduino/hardware/pico/rp2040/<files are here> on my system.
2) Standard git, so you need `git pull; git submodule update` should the modules be updated.

closed 2021-04-22 by earlephilhower

reppr, 2021-04-22:
thanks a lot :)