earlephilhower/arduino-pico #121: Tone problems /?request?

anon-noname, 2021-05-07:
Thanks for the work to get support for RP2040 working in Arduino IDE. Very much appreciated. Attached is test code. I am using BSP 1.1.0 on a standard / reference PI PICO. Arduino IDE is 1.8.13 (Windows Store 1.8.42.0)

I am having trouble getting tone to work well on the 2040. There appear to be two problems.
1. Processor crashes (not PIO) when tone is in use. I am able to loop through code 14902 times and then the processor stops.  The test code  does only 4 things.
-read and scale analog input (pot)  
-noTone / set tone to scaled analog value
-increment loop counter
-print values to usb console
After processor code execution stops PIO tone generation continues. Jitter and long/short pulses stop as expected - you get a nice jitter free frequency. USB serial monitor goes off line and the IDE cannot reach the 2040 to upload programs. Power cycle and if you start upload before next crash you can load new code. As loop execution time goes down, the number of times through the loop before crash increases. There is no significant variation in loop count (1) for a given code loop.

2. There appears to be no way to do 'synchronous' updates to the waveform. That is to say, change the frequency on the next transition. As a result any update to tone (all require a noTone) cuts off the waveform presently being generated such that sometimes very short or extra long pulses are generated and there is a lot of jitter in the waveform. Changing my code to update only when there is a change in frequency will not solve problem. Possibly lessen but you will still have the same problem when control value is alternating rapidly between values 1 apart. This will not be audible unless there are stray short and long period pulses as is the case now.

It looks like the hardware has sufficient registers to prevent this problem. How exactly to do this in the Arduino IDE is beyond my ability. I am just getting my feet wet. 

I have another very closely related issue. I would like to use a piezo speaker connected to produce audio from tone. With the caveats above, this works. However, as you might expect, the speaker performs less well at higher frequencies than at very low ones. The volume change across the spectrum is substantial. In an effort to help minimize this, I would like to try narrowing the pulse width (less energy available) at lower frequencies as _part_ of a scheme to level and possibly even control volume. I would try this with PWM but I haven't found handles to adjust frequency of PWM waveform to test this idea.

Thanks again and I will be happy to test or post an issue if I should. First I wanted to make certain I am not missing something obvious to those who have more experience.

[ToneTestAlpha0.0.1.txt](https://github.com/earlephilhower/arduino-pico/files/6438262/ToneTestAlpha0.0.1.txt)

earlephilhower, 2021-05-07:
For (2) and the differing pulse widths, that's not what `tone()` is really about.  You need to look at driving the PWM hardware itself (i..e not w/analogWrite).  Tone is defined as a 50:50 period, so changing duty cycle is out of scope.  For the on-edge switching, that's possible but not done in any of the `tone`s I've seen on ARMs.  The `Servo` class implements the logic needed in the PIO to make it happen, but it's not something that's applicable to the general case (esp. since at low frequencies the update edge may be very many milliseconds.) so not implemented.  If you'd like to throw a PR, though, it seems like a reasonable thing to do.

What are you trying to do with the attached code?  Using `tone()` in that way (changing every ms) is definitely not how tone is normally used...it's not meant to be a real-time oscillator, just a simple square wave for things like buzzers and such.  For example, below 1khz the `delay(1)` isn't even guaranteed to allow a single pulse to complete.

I think you would be better served by using PWM (analogWrite a 50% pin, then in the loop use analogWriteFreq() updates).  At least reduce update frequency to, say, 1/10th or 1/30th of a second, from the 1/1000th it's at now.

For (1), it shouldn't hang the core so you may have found a bug.  Can you make a test case that always fails w/o that analogRead() call?  That way, it can be reproduced on my end and there's no issue of how the pot values change (i.e. you could make the sensorValue do a sawtooth wave, for example, to ensure it changes things).

Thx!

anon-noname, 2021-05-08:
Earl

I understand and agree with you for the most part.  Intended use aside, I wouldn't ever expect to see either behavior from double-buffered hardware function that I 'thought' tone was using. The fact that noTone must be sent seems to me to mess up what would otherwise work very well in simple cases as well as those that are more demanding if it worked as I expected.  I will work to better characterize this for my application.  As for what I am trying to do...

The supplied code is just to show what it takes to make the problem occur. As I said in my initial, I want to generate a tone - that can vary from as low as possible to as high as practical for middle aged humans to hear. The 12Khz (appx) that tone produces should on the platform should be ok. The tone should be relatively pure - I know square = odd harmonics.  Having the update hash mixed in is mostly avoidable by code but it should be completely avoidable with the double buffered architecture. The tone needs to be scaled to an external analog voltage. The updating tone needs to be responsive and vary frequency quickly enough that it is useful as feedback to a listener.  Not my application, but think of a metal detector for ground search. This means less .1 second between update opportunities and limited averaging of the ADC signal in my book. Also updating a dotstar array at the same time but that is not a problem on the platform and out of scope.

As far as eliminating the analog read, I forgot to mention that the same crash happens for this loop:
  noTone(14); 
  tone(14, 1000); // pin number, frequency, duration **(frequency value not changed for 'update')**

It still happens at the same loop count every time. A few more times through the loop but there were fewer instructions... Adding delay of 10, 20 or 100  yields the same result. It just takes correspondingly longer. PIO output still remains after crash in all cases.
It seems that this issue is related to the interaction between noTone and tone or at least setting up of tone more than about 14879 times. 

As far as a PR goes, I may eventually be able to contribute but would be surprised if I will be able to help any time soon.  I will work some on this because tone will work for my application Remember, feet are barely wet. BTW, I intended to post this as a discussion thread - OK if you moved it...

earlephilhower, 2021-05-08:
Thanks for the info about the loop that fails.  I'll try a simple test like:
````
void setup() {
   pinMode(14, OUTPUT);
}
void loop() {
  noTone(14);
  tone(14, 1000);
  delay(1);
}
````
And see what I can see.  Thanks to the SWD, it is possible to look at things with GDB when a bad thing happens.

earlephilhower, 2021-05-08:
For posterity:

````
void setup() {
 pinMode(14, OUTPUT);
 pinMode(LED_BUILTIN, OUTPUT);
}
int c=0;
bool x = false;
void loop() {
  noTone(14);
  tone(14, 1000);
  delay(1);
  if ((c++) % 400 == 0) {
    x = !x;
    digitalWrite(LED_BUILTIN, x?HIGH:LOW);
  }
}
````
Fails after a few seconds as the LED stops blinking.  Minimizes the places it could be going bad...

earlephilhower, 2021-05-08:
Release 1.2.1 has the tone fix in it.

Andy2No, 2021-05-14:
- "control value is alternating rapidly between values 1 apart"

That could just be noise on the analog input.  You may need to set a threshold, to detect a change - only update the output when the new value is above or below the old value by at least the threshold level:

if(abs(new_val - lastgood) >= threshold) 
{ lastgood= new_val; ...

You might also want to update if the new value stays constant for a suitable number of loops but doesn't reach the threshold.  That can happen if you're using a pot to supply the voltage, for example - the end points of travel are likely to be stable.

I have a sketch that uses that idea to display a pot value on a 7 segment LED display.  Let me know if you need it.  A threshold of one or two should be enough.

commit d776ec2f (2021-05-08): Fix memory leak in tone/noTone calls (#125)

Was never deleting the tone object on noTone, so eventually the leaked
objects would fill up heap and cause a crash.

Fixes #121
commit f4f2455f (2021-06-03): Allow continuous tone frequency changes

Fixes #121
Supersedes #185

Redo the PIO program to allow the tone generator on a pin to be updated
without interruption, at waveform boundaries.  This allows for things like
sirens or slurs to be implemented simply.

Use an alarm, not the PIO hardware, to manage time-limited tones().

Add a simple siren example.
commit 5a54bc8c (2021-06-03): Allow continuous tone frequency changes

Fixes #121
Supersedes #185

Redo the PIO program to allow the tone generator on a pin to be updated
without interruption, at waveform boundaries.  This allows for things like
sirens or slurs to be implemented simply.

Use an alarm, not the PIO hardware, to manage time-limited tones().

Add a simple siren example.
PR #185 (2021-06-04): fix tone mutex
this pr fix mutex issue when calling tone() on the same pin with different frequency/duration e.g
- tone(pin, 100, 1000) then
- tone(pin, 100, 2000) it will then call noTone(pin) first before create the new Tone object

The issue is mutex is acquired before calling noTone() causing the mutex failed on noTone() therefore the object is never release. This will cause memory leak in the long run and eventually crash the mcu. Therefore fix #121 

Note: this Pr also add ToneHappyBirth to play the tune on the buzzer that some board is equipped with. This can help to quickly notice the  [... cut]
closed 2021-06-04 by earlephilhower