earlephilhower/arduino-pico #133: Boards manager update - Access Denied (Win 7)

Andy2No, 2021-05-14:
I recently installed ArduinoPico, through the Boards Manager, after providing .json the link, in Preferences.

Today, I tried to update to 1.2.2 through the Boards Manager, and got the following error, saying Access Denied.

I'm using Arduino 1.8.13

[image]

Andy2No, 2021-05-14:
After that, ArduinoPico was no longer in the list for "Updatable", so I searched for Pico, which showed it as though it had never been installed - so I chose Install, which appears to have worked, or at least, didn't produce an error:

[image]

Andy2No, 2021-05-14:
It seems I can no longer compile things for the Pico, after performing this update.

I tried to compile and upload this example:
https://github.com/earlephilhower/arduino-pico/tree/master/libraries/rp2040/examples/Multicore

but got this error:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=2E8A_000A -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_757744 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830231 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=2E8A_000A -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_757744 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830231 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.2
Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.2
Detecting libraries used...
[... cut ...] Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.2\\cores\\rp2040" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.2\\variants\\rpipico" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_757744\\sketch\\Multicore.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Raspberry Pi Pico.

Andy2No, 2021-05-14:
I get the same problem after installing 1.2.1.

I also had a problem installing the different version, which I assume is due to file permissions since it complained that a file already exists.  I found a workaround was to Remove the support for the board, then install the version I wanted.

However, I still get this error under 1.2.1:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.1
Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.1
[... cut ...] Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.1\\cores\\rp2040" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.1\\variants\\rpipico" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_470261\\sketch\\Multicore.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Raspberry Pi Pico.

Andy2No, 2021-05-14:
Same problem with 1.2.0:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.0
Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.0
Detecting libraries used...
"C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++" -c -Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\ [... cut ...] Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.0\\cores\\rp2040" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.2.0\\variants\\rpipico" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_470261\\sketch\\Multicore.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Raspberry Pi Pico.

Andy2No, 2021-05-14:
Attempting to revert to 1.1.0, which seems to be what I had before, I still get the same error.  So. possibly this is all just due to Boards Manager installation failing to work properly, due to file permissions?

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_470261 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830112 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino
Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.1.0
Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.1.0
Detecting libraries used...
[... cut ...] Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.1.0\\cores\\rp2040" "-IC:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\1.1.0\\variants\\rpipico" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_470261\\sketch\\Multicore.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Raspberry Pi Pico.

earlephilhower, 2021-05-14:
Sorry, looks like your installation got corrupted.   The actual toolchain files (the G++ complier/etc.) haven't changed in many releases, but it seems somehow either your downloads got corrupted or the expansion of the ZIPs did.  We're not getting reports of this happening to others, so it's not likely a real core issue.

Your best bet it to remove all traces of the RP2040 stuff and restart.  Exit the IDE, first.

I won't have exact spots for Windows, but your logs show files in "C:\Users\andy\AppData\Local\Arduino15\packages\rp2040".  I'd delete everything in the rp2040 directory, and see if you can uninstall/reinstall from the IDE.

Alternatively, check the README for info on how to use a `git clone` version which you would manually install and use, and which would allow you to always be on the bleeding edge if you wanted.

Good luck!

closed 2021-05-14 by earlephilhower

Andy2No, 2021-05-15:
Arduino: 1.8.13 (Windows 7), Board: "Raspberry Pi Pico, 2MB (no FS), 125 MHz, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=2E8A_000A -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_757744 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830231 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\andy\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\andy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=rp2040:rp2040:rpipico:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None -vid-pid=2E8A_000A -ide-version=10813 -build-path C:\Users\andy\AppData\Local\Temp\arduino_build_757744 -warnings=none -build-cache C:\Users\andy\AppData\Local\Temp\arduino_cache_830231 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-mklittlefs-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-openocd-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-pioasm-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-elf2uf2-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-python3.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-gcc.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -prefs=runtime.tools.pqt-gcc-1.1.0-a-81a1771.path=C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771 -verbose D:\Documents\Arduino\arduino-pico\examples\Multicore\Multicore.ino Using board 'rpipico' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.2 Using core 'rp2040' from platform in folder: C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.2.2 Detecting libraries used... "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++" -c -Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-s [... cut ...]
exec: "C:\\Users\\andy\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.1.0-a-81a1771/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Raspberry Pi Pico.

I'm not able to delete C:\Users\andy\AppData\Local\Arduino15\packages\rp2040 or any of it's sub folders, though I can delete individual files.  I appear to have permission to - I just can't do it.  The folders don't get deleted when I try to delete them.

Are there a lot of other people using Windows 7?  I was under the impression this was a new release, so I wouldn't have thought too many people would have encountered the problem, yet.

Andy2No, 2021-05-15:
Correction - I can't delete the .exe files.  Windows appears to let me, and removes them from the Explorer window, but if I refresh it, they're still there.  Somehow, I'm blocked from deleting them or taking ownership.

Andy2No, 2021-05-15:
Okay, I've found out why, and roughly want to do about it.  This may affect other Windows users, because it's caused by MalwareBytes Premium, locking the .exe files used by the compiler.

Symptoms:
Access denied during update in the boards manager - affects the .exe files, because MalwareBytes has locked them.

Access denied during compilation, to one of the .exe files - same reason.

Can't delete the (3?) .exe files - they're locked by MalwareBytes.

So,

Workaround:

1) In MBAM Settings, click the Exclusions tab.  Add an exception for the equivalent of this folder path:

C:\Users\{YOUR_USERNAME_HERE}\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771

- at that stage (because the .exe files were locked) I couldn't exclude the .exe files separately, because I didn't have permission to open them.  Adding separate exceptions instead, and removing this folder exception, might be a better idea - once it's working.

2) Reboot to unlock the files.

3) Do the boards manager installation / upgrade again.

4) Set the board type, e.g. to Raspberry Pi Pico and check it can compile.

earlephilhower, 2021-05-15:
Good detective work!  Antivirus is always funky under Windows, and it seems like MWB is even more of a pain than others.

Andy2No, 2021-05-15:
Thanks.  I don't normally have any problems with it, that I'm aware of, but googling the problem with file locking, it seems to have happened to other people quite a lot.

I can see how an AV could treat the boards manager as suspicious, when it's installing .exe files.  There was no indication of it detecting a problem though.

I'll edit down my posts above, to remove some of the output from the Arduion IDE.  It seems only the last part was ever relevant.

earlephilhower, 2021-05-15:
(I'm glad I'm not the only one to mistype the core name as `Arduion` :laughing: )

If you have `git` you could do a PR and add a note in the installation docs about it.  OTW, I can maybe take your link https://github.com/earlephilhower/arduino-pico/issues/133#issuecomment-841697966 and add it myself.

But, since you did the hard work to figure it out, no reason for you not to get credit in the `git blame`s.

Andy2No, 2021-05-15:
I try not to spell the same word the same way too often.  It shows a lack of imagination :)

I've no idea how to do a PR, but I could look into it.  On the other hand, I'm happy to remain blameless, if you want to add it yourself.

So far, I've only ever used git clone, from the command prompt, following instructions.

earlephilhower, 2021-05-15:
OK, I've added your comments to the docs.  The online help will update automatically and it'll be preserved for posterity...

Andy2No, 2021-05-15:
Excellent.  Thanks :)

Andy2No, 2021-05-16:
I may have declared victory too soon...

Trying to upgrade to 1.3.1, and then to 1.4.0 with the boards manager. I got the same access denied error for 

C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.1.0-a-81a1771\libexec\gcc\arm-none-eabi\10.2.0\cc1plus.exe

despite having an exclusion for it in Malwarebytes.

I got around it this time by just rebooting, then starting the Arduino IDE immediately afterwards, and doing the installation from the boards manager again.  So, possibly that's all that's needed.

Having said that, I can't get 1.4.0 to compile anything (see #140) due to a missing stdint.h, so maybe there are further complications caused by a partial installation or removal that fails with an error.  I now get an error trying to remove 1.4.0:

java.lang.NullPointerException

so I  may need a few more reboot cycles - reboot, remove, reboot, install, perhaps.

I'll experiment a bit more.  Sorry for the less than definite answer.

Andy2No, 2021-05-16:
Okay, the solution this time didn't require any more reboots, which saved some time.

I couldn't "Remove", or install a different release, in the boards manager because I got the java.lang.NullPointerException error each time.

I was able to delete the package folder;
 
C:\Users\andy\AppData\Local\Arduino15\packages\rp2040

without any problems with locked files.

After starting the Arduino IDE, the Boards manager showed Raspberry Pi Pico / RP2040 as not installed, so I installed 1.4.0 again - with no errors, and sketches now compile okay again.

Maybe it would be worth adding the stage of deleting the rp2040 folder to the documentation for this problem... or maybe more details have yet to be revealed, in attempting more updates.   That does seem like a fair bet for a useful step though.