earlephilhower/arduino-pico #134: W5x00 Ethernet libraries not working on this core

khoih-prog, 2021-05-15:
### Describe the bug

Ethernet (Ethernet, Ethernet2, Ethernet3, EthernetLarge) libraries not working with this core, but working OK with [ArduinoCore-mbed RP2040 core v2.0.0](https://github.com/arduino/ArduinoCore-mbed).
It seems that SPI bus is not working correctly and the board can't communicate with the W5500 chip, verified by turning debug output of [w5100.cpp](https://github.com/arduino-libraries/Ethernet/blob/master/src/utility/w5100.cpp) 

### Steps to Reproduce

Using this MRE sketch

```
#include <SPI.h>
#include <Ethernet.h>

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
EthernetClient client;

void setup() 
{ 
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial);

  Serial.println("Ethernet Test");
  
  Serial.print("MOSI:"); Serial.println(MOSI);
  Serial.print("MISO:"); Serial.println(MISO);
  Serial.print("SCK:"); Serial.println(SCK);
  Serial.print("SS:"); Serial.println(SS);

  Ethernet.init(SS);
  
  // start the Ethernet connection:
  Ethernet.begin(mac);
  Serial.print("IP = ");
  Serial.println(Ethernet.localIP());
}

void loop() 
{
}
```

### Expected behavior

Working OK as [ArduinoCore-mbed RP2040 core v2.0.0](https://github.com/arduino/ArduinoCore-mbed).
Connection is according to what shown in Terminal Output (MOSI:3, MISO:4, SCK:2, SS:5)

```
Ethernet Test
MOSI:3
MISO:4
SCK:2
SS:5
IP = 192.168.2.108
```

See more results at [Advanced Webserver](https://github.com/khoih-prog/EthernetWebServer#9-advancedwebserver-on-mbed-raspberry_pi_pico-with-w5x00-using-ethernet3-library)

### Actual behavior

Connection is according to what shown in Terminal Output (MOSI:15, MISO:12, SCK:14, SS:13)

```
Ethernet Test
MOSI:15
MISO:12
SCK:14
SS:13
IP = 0.0.0.0
```

### Information

Arduino 1.8.13
Board: RASPBERRY_PI_PICO
Test using both [arduino-pico core v1.2.1 and v1.2.2](https://github.com/earlephilhower/arduino-pico)

---

Thanks for your amazing core.

Best Regards,

earlephilhower, 2021-05-15:
I don't have a W500 NIC hat to try with, but FWIW the SPI library has been used by many other folks thru the Adafruit GFX libraries and in SDFat.  It is a very thin wrapper around the SDK, so in theory there's not much to go wrong.

It['s possible I've broken something, of course!

I'll try and re-hookup a SD card reader because that required both read and write over SPI to function and see if I can reproduce a failure, but I think we're gonna need your extensive help here to figure this out.

First, just to be clear, those MISO/MOSI/etc. pins are GPX, not the pin # itself (i.e. MOSI == GP15 == pin 20 on the carrier board.).  You probably already have that, but if you could just double-check the wiring?

Next, you can enable SPI debugging in the menus and capture the logs it dumps.  That will show what's being called and what data is being read back.

Also, if you have a simple logic analyzer, can you capture any waveforms on the SPI bus during the run?  That plus the logs would show if, for example, a byte got skipped or re-transmitted  on the wire.

earlephilhower, 2021-05-15:
Oh, and per the docs there are a few unsupported SPI interrupt-info calls.  See the docs for the list:  https://arduino-pico.readthedocs.io/en/latest/spi.html .  If the W500 library needs those, then we need to see what exactly it's trying to do and how to map that to the SDK way of running the SPI ports.

earlephilhower, 2021-05-15:
Ooops, looks like I messed up the default SPI1/SPI0 pinouts.  See page 4 of https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf ....
MISO -> 19
MOSI -> 16
SCK -> 18
CS -> 17

I'll push an update....

earlephilhower, 2021-05-15:
Can you retry with git head and reopen if there's still a problem, @khoih-prog ?

closed 2021-05-15 by earlephilhower

khoih-prog, 2021-05-15:
Yes, I'm working on this now.

khoih-prog, 2021-05-15:
Wow, it's working now. Thanks so much. Will have many more of my libraries to support this core, which is much better (with LittleFS) than current Arduino mbed core

```
W5100 init, using SS_PIN_DEFAULT = 13, new ss_pin = 10, W5100Class::ss_pin = 17
W5100.cpp: detect W5200 chip
Wiznet soft reset
mr=0
W5100.cpp: detect W5500 chip
Wiznet soft reset
mr=0
version=4
chip is W5500
W5100::init: W5500, SSIZE =4096
IP = 192.168.2.108

```

earlephilhower, 2021-05-15:
Great!  I'll drop a new release to get it out there.

khoih-prog, 2021-05-15:
Marvellous :1st_place_medal:

khoih-prog, 2021-05-16:
Testing using [**EthernetWebServer**](https://github.com/khoih-prog/EthernetWebServer) OK now

[image]

OptifySudarshanPatil, 2023-05-20:
I followed instruction from https://arduino-pico.readthedocs.io/en/latest/platformio.html
Still not able to pinpoint and rectify this issue.
All files seems present on location.
[image]

JAndrassy, 2023-05-20:
@OptifySudarshanPatil you have a very different problem.

OptifySudarshanPatil, 2023-05-20:
> @OptifySudarshanPatil you have a very different problem.
@JAndrassy  That could be true,
This is my platformio.ini file and i'm trying to compile "AdvancedWebServer_RP2040_SPI1" for "w5500-evb-pico" on windows 10
What am i missing?

``` 
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower

lib_deps = 
    https://github.com/arduino-libraries/ArduinoHttpClient.git
    https://github.com/khoih-prog/EthernetWebServer.git

```

PR #135 (2021-05-15): Fix SPI default pinouts to SPI0, not SPI1 pins
Should fix #134 .  The SPI pin macros were set using the SPI1 set, not
the SPI0 set of pins it should have been using.
commit 9725155f (2021-05-15): Fix SPI default pinouts to SPI0, not SPI1 pins (#135)

Should fix #134 .  The SPI pin macros were set using the SPI1 set, not
the SPI0 set of pins it should have been using.