earlephilhower/arduino-pico #140: Release 1.4.0 compilation errors - stdint.h

Andy2No, 2021-05-16:
Trying to compile sketches with the 1.4.0 release, for the Raspberry Pi Pico I get an error, which doesn't seem to be related to the problems I've had with installing updates from the boards manager, but I suppose it could be...

The last  part of the output says:

C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.4.0\cores\rp2040/Arduino.h:24:10: fatal error: stdint.h: No such file or directory    24 | #include <stdint.h>       |          ^~~~~~~~~~ compilation terminated. exit status 1 Error compiling for board Raspberry Pi Pico. 

The C:\Users\andy\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.4.0\cores\rp2040 folder looks like it's probably okay - there's at least a couple of dozen .h and .cpp files, but no stdint.h

earlephilhower, 2021-05-16:
Sorry, you still have a toolchain problem.  `stdint.h` is a generic C header that's part of GCC, not this core release.  Maybe MalwareBytes has flubbed things up again?

Andy2No, 2021-05-16:
Ah.  Okay.  I'll try more cycles of virtually turning it off and back on again, later.

Andy2No, 2021-05-16:
I've got it compiling again.  I'll update #133 with how, and close this one.

Thanks.

closed 2021-05-16 by Andy2No