earlephilhower/arduino-pico #160: tinyusb on 1.5.1

artouste, 2021-05-22:
Hello
I am trying to use version 1.5.1 with tinyusb ADAFRUIT support
especially the midi_test example.
normally all dependencies are up to date.
But despite several installations I still get a compilation in error.
It is not yet fully worn?

2 links 
log compil https://www.cjoint.com/doc/21_05/KEwkdRxMi3Y_logpico2.txt
hard copy https://www.cjoint.com/doc/21_05/KEwkr3I674Y_compilpico1.jpg

thank you for your work

earlephilhower, 2021-05-22:
Make sure you have selected `Tools->USB Stack->Adafruit Stack` before trying any of the TinyUSB examples.

artouste, 2021-05-22:
thank you very much indeed I had not seen this point before.
 I will be able to play now with midi and pico

earlephilhower, 2021-05-22:
Good to hear it's working!

closed 2021-05-22 by earlephilhower