earlephilhower/arduino-pico #167: compile error when using Serial with USB stack set to TinyUSB

skybrian, 2021-05-25:
I get a compile error if I print to Serial with TinyUSB enabled:

    //#include <Adafruit_TinyUSB.h>
        
    void setup() {
          pinMode(LED_BUILTIN, OUTPUT);
          Serial.begin(115200);
          while (!Serial) {}
          Serial.println("Ready.");
    }
        
    void loop() {
          digitalWrite(LED_BUILTIN, HIGH);
          delay(1000);
          digitalWrite(LED_BUILTIN, LOW);
          delay(1000);
    }

Here are the errors:

    sketch_may24f.ino.cpp.o: in function `setup':
    sketch_may24f.ino:5: undefined reference to `_ZN17Adafruit_USBD_CDC5beginEm'
    sketch_may24f.ino:6: undefined reference to `_ZN17Adafruit_USBD_CDCcvbEv'
    sketch_may24f.ino:8: undefined reference to `Serial'

The workaround is easy: include the TinyUSB header file in your sketch (uncomment the include).

hathach, 2021-05-25:
It is a must, Serial is based on USB CDC which is also provided by TinyUSB library as well. Therefore you need to include the library header for it to compile and link. For one code fit all you should use

```C
#ifdef USE_TINYUSB
#include <Adafruit_TinyUSB.h>
#endif
```

earlephilhower, 2021-05-25:
I think it may be best just to put that ifdef inside `Arduino.h` to simplify things.  I was going to put a PR in later today.

earlephilhower, 2021-05-26:
Arduino won't scan libraries if they're called from the cores/... folder, as you said.  The only option would be to move the library into cores/rp2040/... and then bracket every single source file with `#ifdef USE_TINYUSB` which is not really acceptable.

Thanks @skybrian  for #168 doc update!

hathach, 2021-05-26:
> Arduino won't scan libraries if they're called from the cores/... folder, as you said. The only option would be to move the library into cores/rp2040/... and then bracket every single source file with `#ifdef USE_TINYUSB` which is not really acceptable.
> 
> Thanks @skybrian for #168 doc update!

one of the tip is just add it to Wire and SPI and other commonly used libraries within the included cores, since those are commonly used, it will trigger arduino to link the library in most sketches 

https://github.com/adafruit/ArduinoCore-samd/blob/master/libraries/SPI/SPI.cpp#L25

earlephilhower, 2021-05-26:
That's quite a hack you found, @hathach  :)  I'm reopening this to see if I can get something like that running here.

hathach, 2021-05-27:
> That's quite a hack you found, @hathach :) I'm reopening this to see if I can get something like that running here.

Got to find way to get user happy :smile:

skybrian, 2021-05-27:
My experiment worked, so this user is already happy. :)

Just doing my bit to smooth the way for the next person.

closed 2021-05-29 by earlephilhower

jfrey-xx, 2023-02-20:
Hello,

@earlephilhower, could it be that this bug resurfaces with 2.7.3? I just upgraded (skipping 2.7.2) and I have the exact same symptoms (and workaround) as described in the first post.

reubenbijl, 2023-10-01:
> Hello,
> 
> @earlephilhower, could it be that this bug resurfaces with 2.7.3? I just upgraded (skipping 2.7.2) and I have the exact same symptoms (and workaround) as described in the first post.

Did you work this out?

PR #180 (2021-05-29): Add TinyUSB include to all libs
Fixes #167

For Serial when selecting TinyUSB.  Can't include in the core because Arduino IDE
will not link in libraries called from the core.  Instead, add the header to all
the standard libraries in the hope it will still catch some user cases where they
use these libraries.

See https://github.com/earlephilhower/arduino-pico/issues/167#issuecomment-848622174
commit 1fd0b040 (2021-05-29): Add TinyUSB include to all libs (#180)

Fixes #167

For Serial when selecting TinyUSB.  Can't include in the core because Arduino IDE
will not link in libraries called from the core.  Instead, add the header to all
the standard libraries in the hope it will still catch some user cases where they
use these libraries.

See https://github.com/earlephilhower/arduino-pico/issues/167#issuecomment-848622174