earlephilhower/arduino-pico #169: i2c: Wire only works with some pins. (Need to use Wire1 instead.)

skybrian, 2021-05-26:
If you use setSDA() and setSCL() on Wire then it apparently only works for pins labeled "I2C0" on the Pico [pinout diagram](https://datasheets.raspberrypi.org/pico/Pico-R3-A4-Pinout.pdf). I had the bad luck of choosing pins labeled "I2C1" and it silently didn't work. For those pins you need to use Wire1.

This seems awkward, but if it can't be helped, maybe stating this more explicitly on the [i2c page](https://arduino-pico.readthedocs.io/en/latest/wire.html) might help others avoid this gotcha?

earlephilhower, 2021-05-26:
> ...silently didn't work...

Crud, I **thought** I had added warnings when a `setXXX()` failed when in debug mode, but it looks like I didn't get around to it.

It's just the way the Pico HW muxes are set up.  Only certain pins are allowed for each peripheral copy (I2C, SPI, and UART) as well as per actual pinout (i.e. you can't swap SDA/SCL pins).  You really need a copy of the Pico pinout sheet like you linked.

earlephilhower, 2021-05-26:
My preference is to keep them both separate and usable.  The thing is, you're limited to specific pins anyway (i.e. can't ever do SCK on GPIO0), so there is always some knowledge needed.  Exposing both HW sets lets you do things like run 2 separate apps doing I2C (maybe some kind of safety critical thing where you read some data from one I2C bus, compute on it, then fire it to the other I2C bus to be acted upon).

For now, even DEBUG mode doesn't warn you when the set call fails (it does return `false` but that's pretty much ignored).

I think the right thing to do, then would be to do a complete `panic()` (i.e. dump and halt) in all modes, debug or not.  I'll throw in a PR to that effect for the 3 peripherals later tonight.

PR #182 (2021-05-29): PANIC on attempting an invalid setXXX pin
Fixes #169

Trying to change pinout while running, or to an illegal configuration,
will now immediately panic() with an error message.  Such an attempt
is a pretty big problem since pinouts are hardware related/static.

Prior code would fail silently and return false, but nobody checked
the setXXX return values, anyway.
commit fd685aac (2021-05-29): PANIC on attempting an invalid setXXX pin (#182)

Fixes #169

Trying to change pinout while running, or to an illegal configuration,
will now immediately panic() with an error message.  Such an attempt
is a pretty big problem since pinouts are hardware related/static.

Prior code would fail silently and return false, but nobody checked
the setXXX return values, anyway.
closed 2021-05-29 by earlephilhower