earlephilhower/arduino-pico #171: ArduinoCore-API potential symblink issue on windows

hathach, 2021-05-26:
Although I have no issues with this on Linux, windows may have issue with symlik. Would it be better just include the core api as submodule directly into core/api instead of using symlink i.e change from

```
[submodule "ArduinoCore-API"]	
path = ArduinoCore-API
url = https://github.com/earlephilhower/ArduinoCore-API.git
```

to 

```
[submodule "ArduinoCore-API"]	
path = cores/rp2040/api
url = https://github.com/earlephilhower/ArduinoCore-API.git
```

earlephilhower, 2021-05-26:
Thanks, but the symlink is actually to a subdirectory of the ArduinoCore-API repo, so moving it to the cores/rp2040 directory would end up with `cores/rp2040/api/api/<header.h>`.  It would also place the ArduinoCore unit tests in the Arduino build path...

That said, I did just try it on a W10 install and the symllink came out messed up (just a text file, not a FS join/union/whatever it's called on NTFS) so you have found a real problem.  At least docs need updating, I guess.

hathach, 2021-05-27:
ah you are right, also since the repo include other sources in test folder, it won't be directly included in the core. What a pain !! maybe update the get tools script to also copy api on windows. I leave it up to you to decide, feel free to close this when you think it is addressed.

earlephilhower, 2021-05-28:
After looking at this and seeing what git can (not) do, I think I will just do a copy of the directory tree.  To keep things safe, I will add a CI job that checks that the copy of the tree and the submodule contents exactly match.  So Win and Linux will be all good, no symlinks will be needed.  The API is pretty stable, anyway.

hathach, 2021-05-28:
Yeah just copy it, we will need to modify it later 😈 😈

PR #176 (2021-05-28): Check in Arduino API directory to avoid symlink
Fixes #171

Under Windows, it is very hard to make a symlink and by default git won't
make one to the Arduino API directory, causing annoying build errors.

Avoid the issue by duplicating the ArduinoAPI directory explicitly and using
CI to verify that there are no differences between the two.
commit ab02801d (2021-05-28): Check in Arduino API directory to avoid symlink (#176)

Fixes #171

Under Windows, it is very hard to make a symlink and by default git won't
make one to the Arduino API directory, causing annoying build errors.

Avoid the issue by duplicating the ArduinoAPI directory explicitly and using
CI to verify that there are no differences between the two.
closed 2021-05-28 by earlephilhower