earlephilhower/arduino-pico #190: Serial Serial1 Serial2 together ?

artouste, 2021-06-06:
Hello
Hello
I'm not sure I understood correctly
if with your port and a pico2040 I can have simultaneously

a serial link via USB <---> PC (comxx port)
and  Serial1 and Serial2 hardware serial

I tried this code which fails to compile (Serial unknown) 

Merci de votre atention
Artouste
-------------------------------------------------
void setup() {
  Serial.begin(9600); // serial by USB
  Serial1.begin(57600); // UART0 TX GPIO0 RX GPIO1 RPI 2040
  Serial.println("DEBUG"); //Serial USB
  Serial2.begin(115200); // UART1 TX GPIO4 RX GPIO5 RPI 2040
}

void loop() {
  // put your main code here, to run repeatedly:

  // Pour simple test  replicateur serial à # vitesses
  if (Serial.available()) {
    byte inByte = Serial.read();
    Serial1.write(inByte);
    Serial2.write(inByte);
  }

  if (Serial1.available()) {
    int inByte = Serial1.read();
    Serial.write(inByte);
    Serial2.write(inByte);
  }
  if (Serial2.available()) {
    int inByte = Serial2.read();
    Serial.write(inByte);
    Serial1.write(inByte);
  }
}

earlephilhower, 2021-06-06:
I just compiled your example and it built just fine.  Make sure you're using `Tools->USB Stack->Pico SDK` or add the appropriate `#include <Adafruit_TinyUSB.h>` is using the Adafruit one.

artouste, 2021-06-06:
thank you for your quick reply
As I am discovering the pico2040 and that I often use MIDI over usb with the "adafruit" stack
I don't have the reflex to think about swapping between the USB stack.
Thanks again for your work
Artouste

closed 2021-06-07 by earlephilhower