earlephilhower/arduino-pico #202: analogWrite() prevents digitalWrite() from working

oerkel47, 2021-06-09:
When using digitalWrite() after using analogWrite() on a gpio the output doesn't change. 

Pair this simple example with an LED for quick visual proof.

`void loop() {

	digitalWrite(20, HIGH);
	delay(2000);
	digitalWrite(20, LOW);
	delay(2000);
	analogWrite(20, 20);
	delay(2000);

}`

earlephilhower, 2021-06-10:
You've got a bug in the sample code (the 2nd `digitalWrite` should write to pin 20), but I was able to repro this.

Basically the analogWrite set an output mux value that digitalWrite never overwrote, so it would always be stuck w/the PWM on the pin (until a `pinMode()` call).  A 1-line fix takes care of it..

closed 2021-06-10 by earlephilhower

oerkel47, 2021-06-10:
Uhh, yep that was just a typo here on github. Thanks for fixing!

PR #203 (2021-06-10): Set GPIO digital output on digitalWrite
Fixes #202
commit 793496da (2021-06-10): Set GPIO digital output on digitalWrite (#203)

Fixes #202