earlephilhower/arduino-pico #208: Arduino nano RP2040 support

Claupio, 2021-06-12:
Please support the nano rp2040 I hate mbedOS and I want to use the pico timer library. It would be easy to support the nina module you have just to declare the nina pins in the core (unfortunatly A6 and A7 are linked to the nina module). I don't use directly a pico only for the Wi-Fi support.

earlephilhower, 2021-06-12:
@Claupio there's a discussion #162 about this, but net result is that I don't have one and there aren't detailed schematics (like Adafruit or RPI or SparkFun provide) so it's a bit complicated.

That said, you *should* be able to build and run right now with this core.  Use the `Tools->Board->Generic RP2040`, then set the `Tools->Flash Size` to the appropriate value.  You then will need to select the `Tools->Boot Stage 2` that corresponds to the onboard flash (most flash chips are incompatible with each other for "reasons").  `Generic SPI/4` is basically guaranteed to work, but will be sloooow.  You can build and try all the `boot stage 2` options, and when one doesn't work you'll just need to reset while holding BOOTSEL to upload another attempt.  Let me know which one works, if you do get one.

At that stage, you should be able to do normal runs like BLINK.INO and MULTICORE.INO.  For the peripherals, I'm not sure how/where the libs are or if they'd be compatible outside of MBED.  Again, if you get anywhere or have a chipe `variant.h` you build that runs do let me know and I'll be happy to add it!

Good luck, and let me know if you get anywhere.

Claupio, 2021-06-14:
I sucsesfully uploaded my schetch in to the nano rp 2040 and I have created a arduino_pins.h for my board. There is only one problem the SPI. I use a TFT Screen 320x240; the screen works well but it recive only the blu color what can be ? (the screen begin function works well and also the set rotation)

Claupio, 2021-06-14:
https://user-images.githubusercontent.com/46087711/121882770-3eff7800-cd11-11eb-84e9-d5329be7a592.mp4

A short video of the screen, the "characters" should be White. I have also problems with the SD Library (the begin function). In my opinion is the same problem of the TFT if i resolve this also the SD will work

Claupio, 2021-06-14:
I have sucsesfully uploaded the schetch, the WiFiNINA module works !!! But the SD library doesn't. So I delete the SD folder in the core libraries and all works. In the file zip there are the contents of the variant folder to add the support for the nano RP2040 (the txt file isn't needed). I will try to support pdm 

[rpconnect.zip](https://github.com/earlephilhower/arduino-pico/files/6648416/rpconnect.zip)

Claupio, 2021-06-14:
The board settings is the same of the raspberry pi pico (I have done a copi paste in the boards.txt (obviusly I changed the board name and the variant folder))

earlephilhower, 2021-06-14:
Release 1.8.4 now has the board listed.  Thx!

Claupio, 2021-06-15:
I have modified the PDM library (only two code lines) to work with this core and I added the pin definitions of the microphone in the pins_arduino.h . In the zip you will find the new pins_arduino.h and the PDM library
[PDM.zip](https://github.com/earlephilhower/arduino-pico/files/6656543/PDM.zip)

earlephilhower, 2021-06-15:
Awesome!  I'll pull it in after work tonight and bump a release out.

earlephilhower, 2021-06-19:
I think we're good here.  If you have add'l libs ported, just let me know.  Thx!

closed 2021-06-19 by earlephilhower