earlephilhower/arduino-pico #217: Problems with newer versions

nsilveri, 2021-06-20:
Hi, 
i upgraded the Pico with the last 1.8.6 version after a month without changing my code, the last version i used was probably 1.5.x where all was working good in my project (rover), after the upgrade nothing works. I downgrade to the 1.5.x version and all comes back to works. What is changed? My rover use I2C0 and HWSerial to receive commands from Raspberry Pi Zero both managed by core0 and the I2C1 managed by core1
Thanks

earlephilhower, 2021-06-20:
You're going to have to be more specific than that to get any help, sorry.

My guess is that the latest versions use the default HW pinouts according to the Rpi Pico datasheet, so  the UART and I2C pins will have moved.  Look at the `Wire.setSDA/Wire.setSCL` calls to move the I2C interface to the old pins, and `Serial1.setRX/Serial1.setTX` for the UART pins.

earlephilhower, 2021-06-26:
No feedback, so closing.  If you can be more specific, and adding the appropriate pinout change calls doesn't help, drop a note.

nsilveri, 2021-06-26:
> No feedback, so closing. If you can be more specific, and adding the appropriate pinout change calls doesn't help, drop a note.

Sorry, i was out this week so i couldn't try to change PINs, however my pin config is this:

  ```
void setup() {
    Serial1.setRX(1);
    Serial1.setTX(0);
    Serial1.begin(9600);
    EEPROM.begin(512);
    scanner.Init();
    Wire.setSDA(4);
    Wire.setSCL(5);
    Wire.begin();
    ...
    }
```

```
void setup1() {

  Wire1.setSDA(2);
  Wire1.setSCL(3);
  Wire1.begin();
  Serial.begin(9600);
  ....
  }
```
i hope this can help you

earlephilhower, 2021-06-26:
No problem!  Reopened.  Just trying to keep the tracker tidy.

Again, though, I'll need some more info as to what exactly isn't working.  Lots of folks are using this (and me, too) and while it's not perfect, of course, it's running pretty well for their peripheral use.

Eyeballing the setup pin configs, they look OK and seem to be executed before the peripheral.begin() call so that's looking good...

nsilveri, 2021-06-26:
Thanks for reopen, i think the problem are not the pin but in another place, is possible that some library became incompatible?
i use this:
```
#include <Wire.h>
#include "SerialUART.h"
#include <HCSR04.h>
#include <EEPROM.h>
#include <Servo.h>
#include <i2cdetect.h>
#include <Adafruit_PWMServoDriver.h>
#include "I2CScanner.h"
#include "Adafruit_VL53L0X.h"
I2CScanner scanner;
Servo radar_servo;
```

i know that Adafruit_PWMServoDriver.h is not fully compatible with Pico but i have not alternative becouse i use 12 servomotors to manage the rover movments, however the pico seems to be fully freezed when i turn it on with newer version.

this is the link with my code [](https://github.com/nsilveri/SilverRover/blob/main/SilverRover.ino)

earlephilhower, 2021-06-26:
Sorry, you've got a lot going on in that sketch.  Not sure what to tell you, other than start commenting things out and see where the real problem lies.

If the Adafruit lib has an issue, I suggest pinging their repo as they are aware of this core and have submitted board definitions for their products here.  If you have a hang, you can always use the PicoProbe and GDB to interrogate the core and see where things are stuck.  The docs in the Rpi Pico manuals are pretty good on getting this going, and GDB and OpenOCD are packaged and installed w/the core.

nsilveri, 2021-06-26:
Ok thanks for the help, unfortunately this period I am busy with the university, when I have time I will try to understand where the problem is, if it is more convenient for you, close the Issue, if I find the cause of the problems I will communicate it to you.

closed 2021-06-26 by nsilveri

nsilveri, 2021-08-08:
Hi, I found the problem, apparently there is some problem with the library https://github.com/luisllamasbinaburo/Arduino-I2CScanner, by removing it everything is back to work even with the latest version (1.9.3)