earlephilhower/arduino-pico #218: compile error: arm-none-eabi-g++: exec format error

pandalanax, 2021-06-22:
Hey i wanted to move to arduino-pico. I do have a picoprobe with another pico.
I was able to build and upload the programs in vscode as described in  Chapter 7 of the [getting-started-with-pico.pdf](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf)
However compiling in arduino ide throws the following error:

```
Arduino: 1.8.15 (Linux), Board: "Raspberry Pi Pico (Picoprobe), 2MB (no FS), 125 MHz, Disabled, None, Pico SDK"

/home/user/Downloads/arduino-1.8.15/arduino-builder -dump-prefs -logger=machine -hardware /home/user/Downloads/arduino-1.8.15/hardware -hardware /home/user/.arduino15/packages -tools /home/user/Downloads/arduino-1.8.15/tools-builder -tools /home/user/Downloads/arduino-1.8.15/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /home/user/Downloads/arduino-1.8.15/libraries -libraries /home/user/Arduino/libraries -fqbn=rp2040:rp2040:rpipicopicoprobe:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None,usbstack=picosdk -vid-pid=2E8A_0004 -ide-version=10815 -build-path /tmp/arduino_build_816255 -warnings=none -build-cache /tmp/arduino_cache_644367 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-python3.path=/home/user/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/home/user/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-pioasm.path=/home/user/.arduino15/packages/rp2040/tools/pqt-pioasm/1.2.0-a-407f016 -prefs=runtime.tools.pqt-pioasm-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-pioasm/1.2.0-a-407f016 -prefs=runtime.tools.pqt-openocd.path=/home/user/.arduino15/packages/rp2040/tools/pqt-openocd/1.2.0-a-407f016 -prefs=runtime.tools.pqt-openocd-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-openocd/1.2.0-a-407f016 -prefs=runtime.tools.pqt-gcc.path=/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016 -prefs=runtime.tools.pqt-gcc-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016 -prefs=runtime.tools.pqt-mklittlefs.path=/home/user/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.2.0-a-407f016 -prefs=runtime.tools.pqt-mklittlefs-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.2.0-a-407f016 -prefs=runtime.tools.pqt-elf2uf2.path=/home/user/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.2.0-a-407f016 -prefs=runtime.tools.pqt-elf2uf2-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.2.0-a-407f016 -verbose /home/user/Downloads/arduino-1.8.15/examples/01.Basics/Blink/Blink.ino
/home/user/Downloads/arduino-1.8.15/arduino-builder -compile -logger=machine -hardware /home/user/Downloads/arduino-1.8.15/hardware -hardware /home/user/.arduino15/packages -tools /home/user/Downloads/arduino-1.8.15/tools-builder -tools /home/user/Downloads/arduino-1.8.15/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /home/user/Downloads/arduino-1.8.15/libraries -libraries /home/user/Arduino/libraries -fqbn=rp2040:rp2040:rpipicopicoprobe:flash=2097152_0,freq=125,dbgport=Disabled,dbglvl=None,usbstack=picosdk -vid-pid=2E8A_0004 -ide-version=10815 -build-path /tmp/arduino_build_816255 -warnings=none -build-cache /tmp/arduino_cache_644367 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-python3.path=/home/user/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=/home/user/.arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed -prefs=runtime.tools.pqt-pioasm.path=/home/user/.arduino15/packages/rp2040/tools/pqt-pioasm/1.2.0-a-407f016 -prefs=runtime.tools.pqt-pioasm-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-pioasm/1.2.0-a-407f016 -prefs=runtime.tools.pqt-openocd.path=/home/user/.arduino15/packages/rp2040/tools/pqt-openocd/1.2.0-a-407f016 -prefs=runtime.tools.pqt-openocd-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-openocd/1.2.0-a-407f016 -prefs=runtime.tools.pqt-gcc.path=/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016 -prefs=runtime.tools.pqt-gcc-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016 -prefs=runtime.tools.pqt-mklittlefs.path=/home/user/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.2.0-a-407f016 -prefs=runtime.tools.pqt-mklittlefs-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-mklittlefs/1.2.0-a-407f016 -prefs=runtime.tools.pqt-elf2uf2.path=/home/user/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.2.0-a-407f016 -prefs=runtime.tools.pqt-elf2uf2-1.2.0-a-407f016.path=/home/user/.arduino15/packages/rp2040/tools/pqt-elf2uf2/1.2.0-a-407f016 -verbose /home/user/Downloads/arduino-1.8.15/examples/01.Basics/Blink/Blink.ino
Using board 'rpipicopicoprobe' from platform in folder: /home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6
Using core 'rp2040' from platform in folder: /home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6
Detecting libraries used...
/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016/bin/arm-none-eabi-g++ -c -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico (Picoprobe)\"" -Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6/ @/home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6/lib/platform_inc.txt -fno-rtti -std=gnu++17 -g -w -x c++ -E -CC -DSERIALUSB_PID=0x000a -DF_CPU=125000000L -DARDUINO=10815 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -I/home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6/cores/rp2040 -I/home/user/.arduino15/packages/rp2040/hardware/rp2040/1.8.6/variants/rpipico /tmp/arduino_build_816255/sketch/Blink.ino.cpp -o /dev/null
fork/exec /home/user/.arduino15/packages/rp2040/tools/pqt-gcc/1.2.0-a-407f016/bin/arm-none-eabi-g++: exec format error
Error compiling for board Raspberry Pi Pico (Picoprobe).

``` 

Thanks

pandalanax, 2021-06-22:
System is `Linux hostname 5.8.0-55-generic #62~20.04.1-Ubuntu SMP Wed Jun 2 08:55:04 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

earlephilhower, 2021-06-22:
Looks like you have an installation issue and somehow one of the tarballs got corrupted.  I'd suggest removing the core from the boards manager and re-installing.  FWIW I run Ubuntu 18.04 and also just tried on a Ubuntu 20.04 VM without incident.

(sorry for image, I'm running 20.04 in virt-manager)
[image]

pandalanax, 2021-06-23:
Cheers, reinstalling the core worked!
Thanks for your work!

closed 2021-06-23 by pandalanax