earlephilhower/arduino-pico #234: analogWriteFreq() does not work as expected

jaszczurtd, 2021-07-01:
Hello there,
Firstly I just wanted to thank you for your great job - this Arduino support for Pico you provided is just amazing!

Anyway, I'm building some hardware for my car project, and some of the engine solenoids requires a relatively low PWM frequencies (20-100Hz max).
I'm using two PWM channels on Pico otputs 9 and 10.
Following the docs, I was trying to change the PWM frequency using analogWriteFreq() with parameter 100, and I was expecting to get 100Hz as base frequency for my both PWM channels.
Unfortunately it seems that analogWriteFreq() does not take any effect, the minimum frequency I'm able to get is something around 1Khz. Is there any chance that I'm doing something wrong, and this can be fixed somehow?

I'm also using the analogWriteResolution() to set desired output resolution for PWM.

Here is the sample of my code:

#define PWM_WRITE_RESOLUTION 9
#define PWM_RESOLUTION 511

[...]
    analogWriteFreq(100);
    analogWriteResolution(PWM_WRITE_RESOLUTION); 
    analogWrite(pin, (PWM_RESOLUTION - val));

Thanks in advance for any hint or help!

Best,
Marcin

earlephilhower, 2021-07-01:
This is a result of the way the PWM is generated.  There's a loop counter (from 1...65535) and a PWM-increment clock division.  The lowest PWM divider is used in this case, producing a clock of (125MHz/XXX).  Then, after 512 ticks the PWM resets and the PWM output goes back to 1.  So period is (125MHZ/XXX)*512.

I'm not sure there is a straightforward way to do better, but for now this points to an easy fix for your problem:  Make your PWM_RESOLUTION 16 bits/65535.

This will also give you the most resolution for solenoid control, too.  You'll have to scale all writes by (65536/512), but you obviously get the possibility of having 128 steps between each of your original steps.

I might be able to do the same thing behind the scenes (scale analogWrites to always be close to 65K), but need to think about it some more.  I have a feeling there are unpleasant corner cases there...

jaszczurtd, 2021-07-02:
Thanks, that indeed solved my problem.
I understand that my case may not be a valid case for everyone, but if you don't feel comfortable about changing that on your side, it might be a good idea to simply mention this dependency in docs, where analogWriteFreq() and analogWriteResolution() functions are mentioned.

Thanks again! :)

earlephilhower, 2021-07-09:
Reopening since I think this is a real issue.  I think in calculating the clock frequency wrong here…

earlephilhower, 2021-07-10:
@jaszczurtd would you be able to undo your changes and give #244 a try with the same config you started with (100hz, 256 scale)?

PR #244 (2021-07-15): Fix PWM frequency at low and high frequencies
The PWM HW can only divide 125MHZ sysclk by 1...256.  That's only down to
about 500khz.  If the max count of the PWM (analogWriteScale) is too low
then the actual PWM period will be much less than desired (and PWM frequency
of course much higher).

For example, at analogWriteFreq(100); analogWriteScale(256); the true
PWM period would be 125M / 256 (pwmdiv) / 256 (scale) = ~2khz.

Conversely, at high frequencies and large scales it is impossible to achieve
the requested frequency and a much lower one would be generated.  For
example: freq(60K), scal [... cut]
commit 05356da2 (2021-07-15): Fix PWM frequency at low and high frequencies (#244)

The PWM HW can only divide 125MHZ sysclk by 1...256.  That's only down to
about 500khz.  If the max count of the PWM (analogWriteScale) is too low
then the actual PWM period will be much less than desired (and PWM frequency
of course much higher).

For example, at analogWriteFreq(100); analogWriteScale(256); the true
PWM period would be 125M / 256 (pwmdiv) / 256 (scale) = ~2khz.

Conversely, at high frequencies and large scales it is impossible to achieve
the requested frequency and a much lower one would be generated.  For
example: freq(60 [... cut]
closed 2021-07-15 by earlephilhower