earlephilhower/arduino-pico #235: DS3231 on RTCLib giving incorrect signals to RP2040

KineticLabs25, 2021-07-03:
- Arduino board:  Adafruit Feather RP2040 with DS3231 featherwing

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.15

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

- Using the "ds3231" example code in RTCLib with the RP2040 feather and earlephilhower RP2040 core, I receive this in the serial monitor: 
[image]

- By switching over to a feather express M0+ with the adafruit SAMD core, and changing nothing within the code I receive this:
[image]

I'm not sure what's causing the issue, the core or the library, but the signal seems to be getting a little confused. Time updates and everything runs with no issue on the M0+, but doesn't seem to be friendly with the RP2040 right now. I would appreciate someone a little smarter than me taking a look!

khoih-prog, 2021-07-03:
Just the issue with the **library**, not the core.

You can try this [**DS323x_Generic** library](https://github.com/khoih-prog/DS323x_Generic) which has been tested OK with RP2040 using this core or Arduino-mned core

1. [Alarm_Ethernet on RASPBERRY_PI_PICO with W5500](https://github.com/khoih-prog/DS323x_Generic#10-alarm_ethernet-on-raspberry_pi_pico-with-w5500)
2. [Alarm_Ethernet on MBED RASPBERRY_PI_PICO with W5500](https://github.com/khoih-prog/DS323x_Generic#9-alarm_ethernet-on-mbed-raspberry_pi_pico-with-w5500)

earlephilhower, 2021-07-09:
I'm unable to reproduce with the latest Adafruit's RTClib https://github.com/adafruit/RTClib which is maintained and current.  I have a feeling youre using I suggest you give them a try as the API is the same AFAICT and it just works  The other libs @khoih-prog mentioned can also be used.

Here's the demo output on my own system:
````
2021/7/9 (Friday) 2:36:38
 since midnight 1/1/1970 = 1625798198s = 18817d
 now + 7d + 12h + 30m + 6s: 2021/7/16 15:6:44
Temperature: 28.00 C

2021/7/9 (Friday) 2:36:41
 since midnight 1/1/1970 = 1625798201s = 18817d
 now + 7d + 12h + 30m + 6s: 2021/7/16 15:6:47
Temperature: 28.00 C

2021/7/9 (Friday) 2:36:44
 since midnight 1/1/1970 = 1625798204s = 18817d
 now + 7d + 12h + 30m + 6s: 2021/7/16 15:6:50
Temperature: 28.00 C
````

earlephilhower, 2021-07-09:
Also, check your wiring.  On the R Pi Pico RP2040 board, SDA = GP4 (pin 6), SCL= GP5 (pin 7).  See https://datasheets.raspberrypi.org/pico/Pico-R3-A4-Pinout.pdf

closed 2021-07-09 by earlephilhower