earlephilhower/arduino-pico #236: Calling a value returning function that doesn't actually return anything crashes the program

RandomHacks-Git, 2021-07-04:
I just spent a few hours trying to figure out what was wrong with my code as it was behaving very strangely (a for loop wouldn't increment) altough it was working just fine with the oficial core as well as with other microcontrollers.

I found I had defined a value returning function with several ifs and returns but the part the code was following didn't actually have a return in the end and therefore the pico was crashing. Here is a very simple example that replicates the issue in case my explanation isn't clear enough. Notice how printTest( ) doesn't return anything but is defined as an int returning function. I can see how a lot of people could get stuck with this issue as it isn't really easy to debug if the code is large.

```
int i;

void setup() {
  Serial.begin(115200);
}

void loop() {
  printTest();
  delay(500);
}

int printTest(){
  Serial.println(i);
  i++;
}
```

earlephilhower, 2021-07-04:
Good catch.  That is a programming error and according to the C spec is "undefined behavior" so crashing is acceptable there.

I was going to port in the following patch from the ESP8266, but completely forgot.  Thanks for reminding me!
https://github.com/esp8266/Arduino/pull/8165

RandomHacks-Git, 2021-07-04:
Yeah it definitely isn't a mistake I usually do but this completely escaped me for a few hours until I finally realised what was going on😅
A patch would be awesome.

earlephilhower, 2021-07-04:
Not only did I need to add the patch, it turns out I had forgot to add any of the warning options to the actual compile options.  D'oh!

No matter what the warning options, your sample will fail to build with the following report using the PR:
````
Compiling sketch...
/home/earle/Arduino/hardware/pico/rp2040/system/arm-none-eabi/bin/arm-none-eabi-g++ -c -Werror=return-type -I/home/earle/Arduino/hardware/pico/rp2040/tools/libpico -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID=0x2e8a -DUSB_PID=0x000a "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefix/home/earle/Arduino/hardware/pico/rp2040/ @/home/earle/Arduino/hardware/pico/rp2040/lib/platform_inc.txt -fno-rtti -std=gnu++17 -g -DSERIALUSB_PID=0x000a -DF_CPU=125000000L -DARDUINO=10815 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -I/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040 -I/home/earle/Arduino/hardware/pico/rp2040/variants/rpipico /tmp/arduino_build_863393/sketch/sketch_jul04a.ino.cpp -o /tmp/arduino_build_863393/sketch/sketch_jul04a.ino.cpp.o
/home/earle/Arduino/sketch_jul04a/sketch_jul04a.ino: In function 'int printTest()':
sketch_jul04a:15:1: error: no return statement in function returning non-void [-Werror=return-type]
   15 | }
      | ^
cc1plus: some warnings being treated as errors
exit status 1
````

PR #237 (2021-07-04): Add warning flags to build, catch missing return
Fixes #236

Compiler warning flags were completely ignored/missed in platform.txt.
Add them, as normal, and include -Werror=return-type because GCC will
produce crashing apps when a function return value is missing.
commit ebd22899 (2021-07-04): Add warning flags to build, catch missing return (#237)

Fixes #236

Compiler warning flags were completely ignored/missed in platform.txt.
Add them, as normal, and include -Werror=return-type because GCC will
produce crashing apps when a function return value is missing.
closed 2021-07-04 by earlephilhower