earlephilhower/arduino-pico #240: Removing COM Port

dillysfriend, 2021-07-06:
I am making a Pico show up as a gamepad using tinyusb. This part is working correctly and I can see the gamepad with button presses in Windows. However, the software I am trying to use the gamepad with appears to be affected by having the COM port on the same device as the gamepad. I was wondering if there is a way to use this library, but not have the COM Port on the Pico? I have looked around the libraries' files, but the COM port appears to be pretty integrated into the library and I'm not really sure where to start. Any help is appreciated. Thanks!

earlephilhower, 2021-07-06:
Are you using the TinyUSB option?  `Tools->USB Stack->Adafruit TinyUSB`  That should not start a COM port w/o you actually writing code to make it happen.

Alternatively, there is a PR open already to disable the default Serial port (but then, of course, you can't do automatic upload and will need to unplug and hold BOOTSEL to upload).

dillysfriend, 2021-07-06:
For whatever reason, the COM port was showing up without my initializing a serial port (I am using a few hardware libraries, so it could be one of them). However, I found Serial.end() on that new PR request and that fixed it for me! Thanks again!

closed 2021-07-06 by dillysfriend

hathach, 2021-07-06:
@earlephilhower Serial.begin() will always be called by TinyUSB init() since I think it is what 90% people would want. Also Serial is required for some mcus (nrf,samd) to enter DFU. For those that dont want Serial, calling Serial.end() would do the trick